MarlinFirmware/Marlin #18648: LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Graphical LCD, or EXTENSIBLE_UI

mclemmon, 2020-07-14:
### Bug Description

DWIN_CREALITY_LCD not compiling correctly. 

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4919216/Marlin.zip)

### Steps to Reproduce

1. Disable current LCD option
2. Enable DWIN_CREALITY_LCD

**Expected behavior:** compile

**Actual behavior:** a bunch of errors about not having an LCD. 

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-07-14:
This setting definitely errors, But is it in error? 
Does the .DWIN_CREALITY_LCD work with this setting?

thinkyhead, 2020-07-15:
The Ender 3 V2 screen code only displays the SD card progress. It is not yet tailored like the other displays to show the progress set by `M73` as an alternative.

ChipCE, 2020-07-16:
@thinkyhead  I have the same problem but with <code>REPRAP_DISCOUNT_SMART_CONTROLLER</code>

ellensp, 2020-07-16:
@ChipCE The REPRAP_DISCOUNT_SMART_CONTROLLER is a character display not a Graphical LCD

closed 2020-07-16 by ellensp

ChipCE, 2020-07-16:
@ellensp  Sorry if I'm wrong. I know it is not the lcd is character display, but is there any reason LCD_SET_PROGRESS_MANUALLY cannot work on the 2004 LCD?  I think in Prusa fw they use M73 R[time] to set the remain time.

ellensp, 2020-07-16:
LCD_progress_bar is the graphical bar. it is not the same as the simple time remaining display. even though it does try and convey the same information.

ChipCE, 2020-07-16:
@ellensp  sorry for my unclear msg. I mean the 2004 LCD should be able to support LCD_SET_PROGRESS_MANUALLY just to set the remain time with M73. Why LCD_PROGRESS_BAR need to be enabled to use LCD_SET_PROGRESS_MANUALLY ?

ellensp, 2020-07-16:
What do you think it should do if you enable LCD_SET_PROGRESS_MANUALLY and then set remaining time with M73?

Technically you don't need LCD_PROGRESS_BAR but you cannot see the value until you turn on the LCD_PROGRESS_BAR feature. And if it didn't tell you to turn it on. you wouldn't know and would wonder why there was no display.

ChipCE, 2020-07-16:
@ellensp  Thank you! I was thinking that option will replace the default timer with the remain time as in the Prusa firmware.