MarlinFirmware/Marlin #18649: Filament runout sensor #4059 - additional firmware options and filtering?

IamGaryGnu, 2020-07-14:
### Description 
Request for filament sensor filtering including time requirements to trigger a positive.  Extension of closed request number 4059.

### Feature Workflow

1. [First Action] - trigger of filament runout
2. [Second Action] - must persist for user defined time, default of either 1 full second or 2 full seconds
3. [and so on...] - if runout maintained for user specified time, raise Z, retract filament 1mm or as appropriate, confirm for another second - and if still positive run through filament run out script - if negative, i.e. reads as filament present within the time frame, it is a false positive, bring z back down, reprime nozzle, and continue

#### Additional Information
Issue occuring for me and other users on the CR-10 S5 - endstop issue occurred with extension cables and possibly remedied by separating endstop wires from loom, and addition of a 104 capacitor across end stops, might be similar possible solution with capacitor or capacitors across the filament sensor, perhaps desensitize the filament out readings.

THANK YOU!
(please forgive if I didn't submit this properly)

italocjs, 2020-07-15:
At least for me software filtering was never so realiable and also added a small measurement error, adding a 100nf cap near the controller board is the best fix possible.  i had to do this on my X endstop on pretty much all printers i assembled because the same cable was delivering power to the stepper and signal to the microswitch.

thinkyhead, 2020-07-15:
This option is not currently exposed in the configurations. But you can define your own `FILAMENT_RUNOUT_THRESHOLD` to change the amount of trigger-time required to produce a runout event. The default value value is 5.

IamGaryGnu, 2020-07-15:
many thanks.  @thinkyhead i didn't know that option exists, thank you.  is the 5 measured in milliseconds or something else? i imagine there is jitter across the switch, so i'd like to set it to be maybe a solid 1 second or even more to trigger - can you advise what value might equate with 1 or even 2 seconds?  again, thank you - i think this will solve my issue, though if anyone is able to comment upon where 1 or 2 capacitors should go on the 3 pin connections that would also be appreciated - thank you!

IamGaryGnu, 2020-07-15:
I see the response on March 5, 2018 on issue 9882 - i'm sorry my searches weren't better - i was looking for filament sensor versus runout.  i see the 5 means 5 consecutive checks - i may change that to 10.  i'd still love to know where to solder the capacitor or capacitors - it's a 3 pin normally closed switch, but unlike the x and y it uses 3 pins.  thanks!

blueeagle69, 2020-07-20:
I take it, you could just use  #define FILAMENT_RUNOUT_DISTANCE_MM if people have smart runout sensors such as the one from Bigtreetech ?

closed 2020-08-26 by github-actions[bot]