MarlinFirmware/Marlin #18650: [BUG] Hotend Idle Timeout only triggers when near minimum temperature

JeremyKennedy, 2020-07-14:
### Bug Description

The Hotend Idle Timeout will only trigger when **near** the minimum temperature, rather than the expected behavior of triggering when **above** the minimum temperature.

### My Configurations

Ender 3 Pro with SKR Mini E3 V2.0

Firmware and configurations: https://github.com/JeremyKennedy/Marlin/tree/main-rebase/Marlin

Relevant settings:
```
#define HOTEND_IDLE_TIMEOUT
#if ENABLED(HOTEND_IDLE_TIMEOUT)
  #define HOTEND_IDLE_TIMEOUT_SEC (5*60)    // (seconds) Time without extruder movement to trigger protection
  #define HOTEND_IDLE_MIN_TRIGGER   175     // (°C) Minimum temperature to enable hotend protection
  #define HOTEND_IDLE_NOZZLE_TARGET   0     // (°C) Safe temperature for the nozzle after timeout
  #define HOTEND_IDLE_BED_TARGET      0     // (°C) Safe temperature for the bed after timeout
#endif
```
### Steps to Reproduce

1. Flash the above firmware. Note that for testing purposes I have the timeout set to 10 (ie. ten seconds).
2. Turn on printer and set the temperature to 185 (ie. ten degrees above the minimum trigger). I used OctoPrint to set the temperature which allows me to see what's happening in a nice graph.
3. Note that the temperature will go up and will not reset, even after waiting several minutes.
4. Set the temperature to 175 (ie. exactly the minimum trigger).
5. The temperature will stabilize around 175, then time out after about ten seconds.

**Expected behavior:** 

The temperature should time out during step 3, when the current temperature is above the minimum trigger.

**Actual behavior:** 

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-07-15:
@italocjs: Was there a reason for doing the check this way instead of checking if the hotend temp was above `HOTEND_IDLE_MIN_TRIGGER`?

italocjs, 2020-07-15:
Hello @thisiskeithb, i mostly gave the idea and a basic code for this feature, @thinkyhead did most of the good coding.  i'm not sure why this has been chosen.

In the original sketch i used ` if ((thermalManager.degHotend(0)) >= FCP_MIN_TRIGGER `  (link here https://github.com/MarlinFirmware/Marlin/pull/16362/commits/b4dbc664226ba2c0f5565bb76f7b39a9a284dd74)

I have been using this first sketch i made since them without any issues.

italocjs, 2020-07-15:
i think the code could work if
`thermalManager.degHotendNear`
is replaced by this
`thermalManager.degHotend(active_extruder)`

but i'm unable to test this for a few weeks (no printers here)

thisiskeithb, 2020-07-15:
That wouldn't compile, but using this worked (tested on my MK3S with a BTT002):

`thermalManager.degHotend(active_extruder) >= HOTEND_IDLE_MIN_TRIGGER)`

I'll put in a PR shortly.

closed 2020-07-15 by thinkyhead