MarlinFirmware/Marlin #18651: [BUG] M48 - probe one too many times

qwewer0, 2020-07-14:
### Bug Description

`M48 - Probe Accuracy Test` seems to probe one too many times,
e.g. `M48 P10` will probe `11` times

### My Configurations

SKR Mini E3 v1.2, BLTouch v3.1
Latest Bugfix-2.0.x 0e1cb109
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4921169/Configuration.zip)

### Steps to Reproduce

1. `G28`
2. `M48 P10`
3. Count it

**Expected behavior:**
`M48 PX` should probe `X` times.

**Actual behavior:**
It probes `X+1` times.

sjasonsmith, 2020-07-14:
I noticed that, but assumed the first one was a homing probe. I didn’t investigate to determine whether that is true.

qwewer0, 2020-07-14:
@sjasonsmith If I count G28's two probing for homing too, then it is still one too many.

thinkyhead, 2020-07-15:
The first probe in `M48` seems to be a test to make sure the probe is functioning properly.
```cpp
// Move to the first point, deploy, and probe
const float t = probe.probe_at_point(probe_pos, raise_after, verbose_level);
bool probing_good = !isnan(t);

if (probing_good) {
  randomSeed(millis());

  LOOP_L_N(n, n_samples) {
    . . .
    // Probe a single point
    sample_set[n] = probe.probe_at_point(probe_pos, raise_after, 0);
    . . .
  }
  . . .
}
```

qwewer0, 2020-07-15:
@thinkyhead Ok, if it is by design then there is no problem here.
Thanks for the information.

closed 2020-07-15 by qwewer0