MarlinFirmware/Marlin #18652: [BUG] delta jerks with m205 x5 y5 z?

FN5969, 2020-07-14:
Bug Description
In marlin 2.0.5.3 on delta, if you use cura 4.6 slicer (i think other slicers too) it cannot properly change jerk.
If you wanna use difrent jerks on difrent printing segments cura generate something like "M205 X10 Y10" and of course marlin change jerk for A and B tower and forget about C tower. And it make no sense. But for some reason marlin give you chanse to have jerks/accel/vel difrent for delta every tower. I think it would be great idea to merge it to ABC jerks/ABC accel/ ABC velocity.

thisiskeithb, 2020-07-14:
Please follow the Issue Template when filing a bug report:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

FN5969, 2020-07-16:
Here is my confings

[Marlin_conf.zip](https://github.com/MarlinFirmware/Marlin/files/4933550/Marlin_conf.zip)

XDA-Bam, 2020-07-18:
It looks like axes X, Y and Z are mapped to towers A, B and C. If so, setting `M205 X10 Y10 Z10` would change jerk to the desired values. Did you check this?

FN5969, 2020-07-18:
Yes It is mapped but it should no be! Becouse m205 x y are directions but not towers motor. And only one thing to change it
- manually. And it's not good way to do. May be some one can fix it?

XDA-Bam, 2020-07-18:
I get that you think  it shouldn't, but how would you address the A-axis jerk individually on a delta, if jerk settings were _not_ mapped X->A, B->Y, etc. ? This is not a simple thing to change - as it is now, it might appear illogical to you, but as best as I can tell, it works as designed.

I think this is more of a slicer problem. The slicer knows this is a delta. It should set the jerk values accordingly in the G-code and right now, it doesn't (slicer oversees the C tower jerk).

FN5969, 2020-07-18:
Thank you mate. May be i wrong but why you can change jerk on delta (or max speed) individualy on separate towers? If this parameters will be merged to one it will be more logic:
VA-Jerk: 10
VB-Jerk: 10
VC-Jerk: 10
to
VABC-Jerk:10 

And it automaticaly fix my problem and some others.

closed 2020-08-24 by github-actions[bot]

thinkyhead, 2020-08-26:
On a Delta the jerk values apply to the movement of the steppers at the towers, and they should all be the same value.

FN5969, 2020-08-26:
But actually it's not doing it, And change only A and B tower. I did some crappy code to fix this problem, but i do not wont show it here (too crappy and fit only delta)