MarlinFirmware/Marlin #18658: inversion of home axis when endstop stuck

maudafaca, 2020-07-15:
I own a geetech gt2560 ver. 1 (with pins for bl touch) and accidentaly my Y ENDSTOP got stuck in ON position. until i descovered the problem my printer started to Y HOME on the oposite direction (Y MAX).
My endstops are the NO REPRAP type.
I think that  the least to expect was that when the endstop is activated is to be warned that the Axis MIN is activated and no movement should be visible. Especialy not to the oposite side.
Am I wrong?

ellensp, 2020-07-15:
This is homing bump in action.

The sequence is:
1) move towards end stop, oh already triggered, so no movement happens.
2) move homing bump amount, moves away from endstop
3) move towards end stop again, oh already triggered, so no movement.

so to you it looks like it is moving away from the end stop. 
It would only move 5mm (default HOMING_BUMP_MM )

thinkyhead, 2020-07-15:
I was thinking that something about the Trinamic drivers is not getting initialized when powering up the board. Perhaps the stepper drivers aren't getting the current, or something else is wonky, leaving them in an indeterminate state.

On first boot from power-off, send `M122` right away to see what comes back.

Then, reset the board and do another `M122` for comparison.

ellensp, 2020-07-15:
no mention of trynamics till tubaian hijacked issue for their issue

AnHardt, 2020-07-15:
A little bit off topic.

As far as i can remember GRBLs homing sequence is a bit different.
First it moves toward the endstop until that triggeres. (does nothing if already triggered).
Then it backs up until the endstop is released.
Than backs up a configurable amount. (Here is home then.)

That has the advantage of never setting home at a point with triggered endstop, even if the configurable amount is zero.

Maybe its time to rethink how #2947 could be realized. But since then it got more complicated due to stalguard-homing.

thinkyhead, 2020-07-15:
Since the OP has lost interest, this particular issue can be closed.

closed 2020-07-15 by thinkyhead