MarlinFirmware/Marlin #18659: [BUG] Ender 3 V2 moving in incorrect axis

jfromell, 2020-07-15:
### Bug Description

On a fresh installation of Marlin-bugfix-2.0 on my Ender 3 V2, when I start a print the printer moves in the Z and X axis as opposed to just laying down filament across the Y axis.

### My Configurations

I used the configuration from Configuration-bugfix-2.0

### Steps to Reproduce

1. Flash firmware with Marlin-bugfix-2.0 using Configuration-bugfix-2.0
2. Start a print
3. Watch the printer move up in the Z-axis and right in the X-axis.

**Expected behavior:** [What you expect to happen]
The printer moves in the Y-axis and lays down filament across the bed

**Actual behavior:** [What actually happens]
The printer moves in the Z and X-axis

ellensp, 2020-07-15:
That might imply there are different motherboards in different v2's   Any chance you can take a clear high res image of your controller?

jfromell, 2020-07-15:
Creality V 4.2.2 it says on the board, i'll try to get a good image but kinda hard with all the cables.

jfromell, 2020-07-15:
[image]

About as good an image as I can get. The controller says "STM32F103".

AnHardt, 2020-07-15:
Is there a massive amount of travel in x an z, or just a little? (compared to the y-move)
Is the printhead near an endstop (soft and/or hard).
What does M119 tell you?
Is or was `SKEW_CORRECTION` defined? If yes - how exactly?
Is or was any kind of bed-leveling involved?
Is any kind of persistent storage involved? Has that been reset to match the configuration? (M50x)

jfromell, 2020-07-15:
Both X and Z travel the full distance, Y doesn't travel at all (this is only during the first line that the starting G-code prints, I've stopped the print after I notice this is wrong)
``SKEW_CORRECTION`` is not defined.
No bed-leveling probe like BLTouch is involved, only manual leveling.
There's no EEPROM access on the Ender 3 V2 at the moment, other than a EEPROM.DAT file that gets generated, probably by the LCD and put on the SD-card.

tvixen, 2020-07-15:
Did it work with Marlin before you upgraded the software, and which version was it.

jfromell, 2020-07-15:
It works just fine with the Creality v1.0.1 firmware, which to my understanding is based on Marlin 2.0.1.

tvixen, 2020-07-15:
Yes i'm sure it did, but did you try Marlin 2.0.5.3 first (before the bugfix version you have uploaded)
If not... and this is your first attempt to run Marlin on your 3dprinter, I suggest you try out Facebook's usergroup for Ender 3.

jfromell, 2020-07-18:
Sorry for late reply, been away for a couple of days. This is my first attempt and first model, so might be a user issue. I will try the code that was pushed a few days ago and see if that fixes things.

fleek, 2020-07-24:
Check if the cables are plugged in correctly.

boelle, 2020-08-23:
@jfromell did you check wires? could you attach your configs?

and re-download [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) again as its updated many times each day

closed 2020-08-30 by github-actions[bot]