MarlinFirmware/Marlin #18660: [BUG] Screen freeze on Ender 2 V2

jfromell, 2020-07-15:
### Bug Description

After flashing the Marlin-bugfix-2.0 firmware to my Ender 3 V2, the screen freezes and I need to take out the SD-card and reboot the machine. When I insert the SD-card again, the screen freezes again. If I delete thee EEPROM.DAT file from the SD-card, the screen works until the next time I remove the SD-card and insert it again.

### My Configurations

I used the Configuration-bugfix-2.0 configurations for Ender 3 V2

### Steps to Reproduce

1. Flash Marlin-bugfix-2.0 firmware
2. Try to navigate the menu
3. Remove the SD-card and delete the EEPROM.DAT file from it
4. Navigate the menu
5. Remove and insert the SD-card
6. Try to navigate the menu

**Expected behavior:** [What you expect to happen]
I can navigate the menu at all times

**Actual behavior:** [What actually happens]
The screen freezes when EEPROM.DAT is present on the SD-card.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tvixen, 2020-07-15:
Why not try Marlin version 2.0.5.3 ?

jfromell, 2020-07-15:
To my knowledge that version doesn't support the Creality V4.2.2 motherboard.

ellensp, 2020-07-15:
It doesn't...   support is only a month ish old.

thisiskeithb, 2020-07-17:
Can you pull a fresh copy of `bugfix-2.0.x` And see if your issues are resolved? Some Ender-3 V2 related code was pushed to Marlin earlier and it may solve your issues.

navickas12, 2020-07-22:
I also have this issue

closed 2020-08-27 by github-actions[bot]