MarlinFirmware/Marlin #18661: [FR] Lock-Screen - not blocking g-codes - just preventing leaving

AnHardt, 2020-07-15:
When using screens with their own processor, having a Marlin emulation mode and having an encoder and all (2-3) cables are connected and a matching display is defined in Marlin,
editing values in TFT-mode with the endcoder causes random actions,
because the encoder is connected to the displays processor and the processor Marlin runs on, in parallel.

When the display is running in its native mode it could send a g-code to send into the lock-screen. Only (randomly) selecting the right number, or the unlock-g-code will let Marlin leave the lock-screen. Random encoder-inputs will not cause random Marlin reactions over the menus because not leaving that screen. But **in contrast** to the lockscreen-feature where is currently worked on this extension should **not block g-code** so values can be purposely be edited over the TFT.
When entering Marlin-mode the TFT could send the unlock-g-code, when entering the TFT-mode the lock-g-code.

thinkyhead, 2020-07-16:
What can we do to determine which mode a TFT is currently in?

AnHardt, 2020-07-16:
Absolutely nothing.
The TFT has to explicitly tell us by sending the lock/unlock-g-codes.
A chicken egg problem.

Tygrys-1, 2020-08-07:
There should also be the escape from the console lock, i.e. very long key press.

closed 2024-02-11 by AnHardt