MarlinFirmware/Marlin #18663: [BUG] Machine doesn't home z when G28 is issued. Debug data contains lots of non ASCII characters

Louis-Cypher, 2020-07-15:
### Bug Description

After updating 2.0.x to the most recent version (pulled yesterday 2020-07-15) homing appears to be broken. Issuing G28 results in homing y, then moving x about 30mm and then homing x (resulting in the probe being somwhere but very likely not over my (pretty small) bed. I didn't change any parameters besides x endstop now being at x max rather than x min. So I changed the endstop to 

```//#define USE_XMIN_PLUG
#define USE_XMAX_PLUG
//#define USE_YMIN_PLUG
#define USE_YMAX_PLUG
#define USE_ZMIN_PLUG
//#define USE_ZMAX_PLUG
```
And the homing directions to 
```#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1
```
[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4925923/marlin.zip)
[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4925927/marlin.zip)

### My Configurations

**Required:** Please find below Configuration Configuration_adv and log file copy pasted from Arduino Serial Monitor

### Steps to Reproduce

1. Acitivate level debugging
2. Set debugging level to max
3. Press autohome or send G28

**Expected behavior:** Homing sequence performed. Debugging data gives meanginful output

**Actual behavior:** x is not homed properly. Debugging output contains a lot of non ascii characters. No meaningful log data.

#### Additional Information

Louis-Cypher, 2020-07-15:
Checked wiring. Wiring OK.

Louis-Cypher, 2020-07-15:
Got it. The endstop was inverted. Hence, homing immediately finds an endstop and hence moves to the predefined homeposition. Debugging output still is scrambled. I personally think that firmware could detect that an endstop is active all the time. Closing anyways since this is not the topic of the issue

closed 2020-07-15 by Louis-Cypher

thinkyhead, 2020-07-16:
> I personally think…

Try the new `DETECT_BROKEN_ENDSTOP` feature.

Louis-Cypher, 2020-07-16:
Thanks! I'll do