MarlinFirmware/Marlin #18664: [FR] Simplify Usage of Configuration.h by having a seperate file with parameters the user changed only

Louis-Cypher, 2020-07-15:
### Description

To me the configuration file nowadays gets more and more difficult to handle and will keep newbies from using this wonderful piece of software Marlin is. It contains much more than just configuration and having almost 2000 lines to me isn't something users should fuzz around with. 

My request would be to have  a seperate Configration_user.h file or the like that carries only the parameters a user changed making upgrading the software from github less cumbersome and time consuming.

### Feature Workflow

1. User copies parameters he wants to change to the Configuration_user.h file
2. Marlin checks for parameters that are set but do not exist at compilation or boot (I think the compiler will complain about non declares variables anyhow, right?
3. During compilation parameters that do exist in both files (Configuration.h and Configuration_user.h) are replaced by Configuration_user.h parameters
3. During Upgrade the user can have the Configuration_user.h file on the gitignore list and doesn't need to merge changes.

#### Additional Information

* I'd be willing to help implementing but my C- coding knowledge is *very rusty*.

thisiskeithb, 2020-07-15:
This isn’t ideal for troubleshooting/maintaining, etc. See the comments from PR https://github.com/MarlinFirmware/Marlin/pull/18632

closed 2020-07-15 by thisiskeithb

Louis-Cypher, 2020-07-15:
I am not at all convinced. 
a) this only is a developers perspective (important, yes; but the majority is users)
b) the comments do not apply to what I suggested (I not at all suggested a template but a file holding only values that are changed from default. People using default values will not nead a file, the majority of users would have a file of about 15-20 lines)
c) I do not have the impression that this suggestion was at all taken into account. It was reviewes by one person and closed with comments that do not apply from my point of view

I would have preferred something I can believe easier, like our focus os different, we have no time for such details or the like.

(over the last almost 10 years of using Marlin I have spent months debugging/ troubleshooting (my own systems) and I *never* had the impression that such a giant configuration file is helpful at all. Configuration should be configuration, nothing more.

Of course this is your decision which I do respect.

thinkyhead, 2020-07-15:
This is a duplicate of various other issues suggesting "better" ideas about configuration. Unfortunately all of them come with compromises and make the maintenance of configurations more complicated. The current two file system is not "perfect" but it is "good enough."

Louis-Cypher, 2020-07-16:
Fair enough. As I said it is just a suggestion and up to the team to decide