MarlinFirmware/Marlin #18665: SKR 1.4 -not compiling when choosing character based LCD

cropduster, 2020-07-15:
[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/4926236/Archiv.zip)

I am running an SKR 1.4 Turbo board ina. Bar converted Prusa MK3 printer. I wanted to merge the latest bugfix 2.0.x changes into my local branches. When trying to compile, I get the following errors:

In file included from Marlin/src/HAL/LPC1768/../../core/../inc/../pins/lpc1769/pins_BTT_SKR_V1_4_TURBO.h:30,
                 from Marlin/src/HAL/LPC1768/../../core/../inc/../pins/pins.h:410,
                 from Marlin/src/HAL/LPC1768/../../core/../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/LPC1768/../../core/serial.h:24,
                 from Marlin/src/HAL/LPC1768/DebugMonitor.cpp:25:
Marlin/src/HAL/LPC1768/../../core/../inc/../pins/lpc1769/../lpc1768/pins_BTT_SKR_V1_4.h:317:6: error: #error "Character LCD not yet supported for SKR 1.4."
  317 |     #error "Character LCD not yet supported for SKR 1.4."
      |      ^~~~~
In file included from Marlin/src/HAL/LPC1768/../../inc/../pins/lpc1769/pins_BTT_SKR_V1_4_TURBO.h:30,
                 from Marlin/src/HAL/LPC1768/../../inc/../pins/pins.h:410,
                 from Marlin/src/HAL/LPC1768/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/LPC1768/HAL.cpp:24:
Marlin/src/HAL/LPC1768/../../inc/../pins/lpc1769/../lpc1768/pins_BTT_SKR_V1_4.h:317:6: error: #error "Character LCD not yet supported for SKR 1.4."
  317 |     #error "Character LCD not yet supported for SKR 1.4."
......

However, when I change the LCD controller to a graphical LCD, the compilation ends flawlessly. 
### My Configurations

Attached is an archive with my config files

### Steps to Reproduce

1. Run compilation with #define REPRAP_DISCOUNT_SMART_CONTROLLER enabled in configuation.h 
**Expected behavior:** Compilation finishes correctly without errors (as before)
**Actual behavior:** Compilation fails with above error message.

#### Additional Information
When disabling #define REPRAP_DISCOUNT_SMART_CONTROLLER and enabling #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER --> all works fine.

GMagician, 2020-07-15:
> #error "Character LCD not yet supported for SKR 1.4."

Self explaining?

sjasonsmith, 2020-07-15:
Based on that, this seems like a pretty simple feature request for someone with the right hardware to address.

cropduster, 2020-07-15:
As I wrote, it worked fine before I did a merge into my local branch. I was running the bugfix 2.0.x from a few weeks before and there was no issue. This was introduced just recently, it seems.

cropduster, 2020-07-15:
I just had 3 conflicting items when merging the latest upstream bugfix 2.0.x into my local branch, one was platform.io, one was configuration_adv.h and one was configuration.h - nothing else and nothing related to LCD.

sjasonsmith, 2020-07-15:
It looks like that was added in response to the following issue:
https://github.com/MarlinFirmware/Marlin/issues/18454

The check might be over-zealous though. The LCD being discussed in that report was a display using a single cable and push-buttons. The display you are discussing uses two cables and an encoder wheel, right?

Given you had this working before and the likelihood that the #error is catching more displays than intended, I'll re-label this is a probable bug.

sjasonsmith, 2020-07-15:
This is the commit related to this issue:
https://github.com/MarlinFirmware/Marlin/commit/d6c177008f465a6369415015bac8d8cfe5101d41#diff-1279ce44430f2e33693efc7f5f2a452c

@ellensp I think it was mostly your comments which led to this check being added. Were those only a concern for the single-cable LCD displays?

cropduster, 2020-07-15:
@sjasonsmith Correct, the display is the two-cable stock Prusa mk3 LCD with the rotary encoder wheel and the push button. It worked fine until recently.

ellensp, 2020-07-16:
@sjasonsmith yes issue is only with displays with resistor network based switches.  HAS_CHARACTER_LCD is way to broad. Should probably be looking for ADC_KEYPAD

thisiskeithb, 2020-07-16:
I can confirm the new warning is too broad and one of my SKR 1.4/2004 character LCD configs no longer builds.

ellensp, 2020-07-16:
That PR should fix it.

thisiskeithb, 2020-07-16:
You're fast. I was just working on one 🙂 I commented on the PR, but I think it should be checking for `HAS_ADC_BUTTONS` based on other code I found.

ellensp, 2020-07-16:
was looking for a 'has'...  didn't see it. will update

thisiskeithb, 2020-07-16:
The sanity check was removed, so I'll close this issue.

closed 2020-07-16 by thisiskeithb