MarlinFirmware/Marlin #18666: [BUG] error: 'STATUS_BED_X' was not declared in this scope; did you mean 'STATUS_FAN_X' when defining STATUS_HEAT_PERCENT

elpeter25, 2020-07-15:
**Marlin Version: 2.0.5.4**

### Bug Description

When uncommenting to define STATUS_HEAT_PERCENT following error appears at compiling time:

```
Marlin\src\lcd\dogm\status_screen_DOGM.cpp: In function 'void _draw_bed_status(bool)':
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:253:30: error: 'STATUS_BED_X' was not declared in this scope; did you mean 'STATUS_FAN_X'?
  253 |           const uint8_t bx = STATUS_BED_X + STATUS_BED_WIDTH;
      |                              ^~~~~~~~~~~~
      |                              STATUS_FAN_X

Compiling .pio\build\LPC1768\src\src\lcd\extui\lib\dgus\DGUSDisplay.cpp.o
*** [.pio\build\LPC1768\src\src\lcd\dogm\status_screen_DOGM.cpp.o] Error 1
```

### My Configurations

[MarlinBug.zip](https://github.com/MarlinFirmware/Marlin/files/4926301/MarlinBug.zip)

### Steps to Reproduce

1. [First Step]
Uncomment line 1287 in Configuration_adv.h:
  //#define STATUS_HEAT_PERCENT       // Show heating in a progress bar
2. [Second Step]
Compile

**Expected behavior:**
Compilation OK

**Actual behavior:**
Compliation error

Thanks so much for making Marlin possible :)

GMagician, 2020-07-15:
Try with bugfix

GhostlyCrowd, 2020-07-15:
I use #define STATUS_HEAT_PERCENT on the latest bug fix and this is not an issue, can you try the latest bug fix? if it fixes the issue, this can be closed if it is still there for you, post your configs for the latest bug fix.

GMagician, 2020-07-15:
Use bugfix but you have to update your configuration files.

Not sure if  SPEAKER is a valid option for your cnf but it seems it generate a compiler error

closed 2020-07-15 by thinkyhead