MarlinFirmware/Marlin #18667: [FR] Seperate chopper timing for each axis

rc-fire, 2020-07-15:
### Description

I would like to tinker my chopper settings as described in #18542, however I use 0.9 degree steppers on X and Y axis, while my Z and E steppers are 1.8 degree. Is it possible to change the chopper timing separately for X, Y, Z and E? 

Or is there currently some workaround to achieve this?

Portzal, 2020-07-15:
@rc-fire this video might help start you in the right direction. https://youtu.be/0OFpiJAMYZQ

thinkyhead, 2020-07-15:
It should be possible to extend the code to have up to 4 different chopper timings. It will require modifications to the settings, `trinamic.cpp`, and `trinamic.h`.

rc-fire, 2020-07-16:
> @rc-fire this video might help start you in the right direction. https://youtu.be/0OFpiJAMYZQ

Ok so according to that I can apply your values only to X and Y with:

` #define TMC_ADV() { \
       stepperX.toff(3);\
       stepperX.hend(-1);\
       stepperX.hstrt(5);\
       stepperY.toff(3);\
       stepperY.hend(-1);\
       stepperY.hstrt(5);\
     }`

It did compile just fine, but my printer is busy right now so I can test it tomorrow. 
Okay if this works this could maybe be closed, but as a feature request probably something like

`#define CHOPPER_TIMING_XY CHOPPER_09STEP_24V
#define CHOPPER_TIMING_Z CHOPPER_DEFAULT_24V
#define CHOPPER_TIMING_E CHOPPER_DEFAULT_24V
`
would be perfect and the most user firendly way.

thisiskeithb, 2020-07-16:
It’s an edge case at this point, but the BigTreeTech GTR + M5 board has up to 11 stepper drivers, so there would probably need to be a lot more options for adjustment/compatibility.

...depending on how far we want to take it.

Argolein, 2020-08-29:
I have a similar setup: X and Y are 0.9 motors, Z are 1.8 motors and E is a 1.8 motor with a geared extruder.
Using the new 0.9 24v chop settings causes the extruder to begin skip steps so I have to use the default 24v chop settings profile for all motors which then causes core noise/vibration for x and y.

So it seems mandatory to chose different chop settings per axis to get rid of this problem.

Camphul, 2020-09-12:
I would also like to see this implement instead of the TMC_ADV workaround. Perhaps use CHOPPER_TIMING as default for all axis' and use axis-specific timings such as X_CHOPPER_TIMING as overrides?

Hyrules, 2020-09-21:
Add my voice to the lot. I'm also in this situation where I have 0.9 Stepper motors on my X Y axis but 1.8 deg on both the extruders and Z motors. It would be nice to be able to set chopping by axis.

thinkyhead, 2020-10-05:
We currently have the option to set Stealthchop separately for XY, Z, and E, so it only allows for three different settings. Would that be granular enough for most purposes?

ChipCE, 2020-10-05:
@thinkyhead That's great. I think it will be enough for most of the machines.

thisiskeithb, 2020-10-18:
Closing as the feature was merged: https://github.com/MarlinFirmware/Marlin/pull/19781

closed 2020-10-18 by thisiskeithb