MarlinFirmware/Marlin #18668: [FR] colder nozzle when unloading the fillament

bohbotjames, 2020-07-15:
### Description

When executing the M702 command, the printer should wait for the nozzle to be colder before extracting the filament.

### Feature Workflow

1. Run the M702 gcode command
2. The nozzle heats up to the temperature requested (example: 230 degrees for PETG)
3. When the temperature is reached, the not end is turned off, and material starts to be fed into the extruder
4. Material keeps being fed into the extruder until the nozzle cools to a certain temperature programmed in the configuration file (example: 215 degrees)
5. When the nozzle is cool enough, the filament is retracted from the extruder to complete the unload command.

#### Additional Information

The way it works at the moment is that when the filament gets extracted, there is stringing at the end of it. By implementing this feature that stringing would be eliminated. Let me know what you think of my idea.

swilkens, 2020-07-15:
What you are suggesting is essentially an automated "cold pull". I like this in principle, but I would do this without the feeding phase.

It would have to be off by default, as this requires tuning. Otherwise we risk setting the temperature too low, potentially causing the extruder to damage the filament.

thisiskeithb, 2020-07-15:
If you tune your `ADVANCED_PAUSE_FEATURE`/`M600` settings correctly, you shouldn't get stringing when doing a filament change (this feature great fine on my various builds).

There's rumors of an upcoming "Maintenance" menu where something like an automated cold pull would likely live.

Liamnerf, 2020-07-31:
you can implement an automatic cold pull like I have using the custom user menus section in the FW.

Marlor, 2020-08-01:
@Liamnerf 
Could you please post your cold pull code?