MarlinFirmware/Marlin #18671: [BUG] The SD card on the LCD FYSETC_MINI_12864 screen cannot be read on the SKR Pro V1.1 motherboard

yangwenxiong, 2020-07-16:
### Bug Description
The SD card on the LCD FYSETC_Mini12864 screen cannot be read on the SKR Pro V1.1 motherboard,
SD card init is OK ，But can not read SD content, I guess ,this is a Bug.

### My Configurations

### Steps to Reproduce

1. #define FYSETC_MINI_12864_1_2  // Configration.h
2. #define RGB_LED                           //Configration.h
3. #define LED_CONTROL_MENU     //Configration_adv.h

**Expected behavior:** 
   Can read SD content
**Actual behavior:** 
   Can not read SD content
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

yangwenxiong, 2020-07-16:
****
[image]
[image]
[image]

ellensp, 2020-07-16:
In your pins files is "#define SDCARD_CONNECTION                  LCD"  but if you set this elsewhere the other definition takes precedence. make sure SDCARD_CONNECTION its disabled in Configuration_adv.h

yangwenxiong, 2020-07-17:
> In your pins files is "#define SDCARD_CONNECTION LCD" but if you set this elsewhere the other definition takes precedence. make sure SDCARD_CONNECTION its disabled in Configuration_adv.h

Hi,The setup you're talking about I'm already disabled

yangwenxiong, 2020-07-17:
> In your pins files is "#define SDCARD_CONNECTION LCD" but if you set this elsewhere the other definition takes precedence. make sure SDCARD_CONNECTION its disabled in Configuration_adv.h

It is possible to read SD CARDS normally on SKR 1.3, I have tried

yangwenxiong, 2020-07-17:
GTR V1.0 only displays the boot screen for about 30 seconds and then no more

sl1pkn07, 2020-07-21:
sorry for the offtopic, but strange you say about fysetc lcd and in the lcd you show in the pictures is a bigtreetech unit (link please?)

boelle, 2020-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

yangwenxiong, 2020-07-25:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4975011/Marlin.zip)

yangwenxiong, 2020-07-25:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4975036/Marlin.zip)

yangwenxiong, 2020-07-31:
The latest version 2.0.6 appears to read the card, but GTRV1.0 does not display

sjasonsmith, 2020-08-17:
I tested on my SKR Pro and it was fine, but as @yangwenxiong said things were very messed up on the GTR.

I've added some GTR changes to a PR I already had open on the SKR Pro. I can now choose in `Configuration_adv.h` whether to use the SD on the display or onboard and they both seem to work properly.

#19047

sjasonsmith, 2020-08-29:
@yangwenxiong based on testing by myself or others I am assuming this is now resolved on both the SKR Pro and the GTR.

closed 2020-08-29 by sjasonsmith