MarlinFirmware/Marlin #18675: Ender 3v2 - Flashing Firmware - Freezes after boot screen status bar

mjmurray68, 2020-07-16:
### Bug Description

After a successful compilation of firmware, flashing fails after boot screen status bar.
**Note: First time Ender 3 user, totally possible this is User Error/Process issue**

### My Configurations

OSX: 10.15.5
Visual Studio Code: 1.47.1
PlatformIO: 4.3.4
Ender 3v2: v1.0.1

Unmodified Configuration files from bugfix-2.0.x for Ender3 v2
platformio.in: modified to build for "STM32F103RET6_creality" board

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Fork Marlin & Configs 
2. Clone to Github Desktop
3. Branch to "myEv2"
4. Copy entire contents from Configurations bugfix2.0.0x Ender-3 V2 directory to Marlin directory
5. Modify "platformio.ini" for correct board in VSCode.
6. Compile (clean)
7. Copy firmware to root of SD card
8. Insert SD Card, and power on Ender

**Expected behavior:** [What you expect to happen]

New firmware would be flashed to Ender.

**Actual behavior:** [What actually happens]

Screen powers on, but remains black as FW is loaded, Splash Screen appears, status bar fills, printer freezes.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4931324/Configs.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-07-16:
This sounds similar to what another user is experiencing in a previously reported issue: https://github.com/MarlinFirmware/Marlin/issues/18675

mjmurray68, 2020-07-16:
@thisiskeithb I was going to check to see if it was a replication, but you referenced this issue #.

thisiskeithb, 2020-07-16:
Oops: https://github.com/MarlinFirmware/Marlin/issues/18660

mjmurray68, 2020-07-16:
I have tried every combination/cycle of flashing, deleting EEPROM.DAT, and rebooting that I can think of. Unlike issue #18660, I am never able to get past the splash screen/progress bar. So while likely related I'm not sure if they're exactly the same.

fleek, 2020-07-17:
I was able to get the system to boot in the screen after reverting 7a0af01e692ddcf2ee55800a035c2276c40bee7c.  However, the encoder wasn't functioning.
 
I think it is related to the EEPROM settings.

thisiskeithb, 2020-07-17:
Take a look at this tweet from @thinkyhead: https://twitter.com/thinkyhead/status/1284171213548531713

Maybe try updating again?

mjmurray68, 2020-07-17:
Thanks everyone. I was waiting on a print to finish, and here's my update:

1. On my existing branch I reverted 7a0af01, built clean, and flashed. The net result was the same: display freezes after progress bar.
2. I deleted my branch, and refreshed as of an hour ago, built clean, and flashed. The net result was the same: display freezes after progress bar.
3. I downloaded the binary from thinkyhead's tweet, and flashed. And it worked perfectly. 
Note: I refreshed the Creality's official v1.0.1 release between each attempt. 

For me it's really about the journey not the destination: I want to successfully build my own branch from scratch, to make sure I have the process nailed down, and then start tweaking things. So while the prebuilt binary works, it's not my end-game, and I'm left wondering if my process is flawed, or if I should just wait on more stable source.

Gurus, should I just close this out?

fleek, 2020-07-17:
Same here, I have added a BLTouch sensor and a filament out sensor to the machine and I need to compile from source to enable the functionalities.  (BLTouch to 5-pin port and filament sensor to z-min-stop)

Since this branch is broken at this point in time, I guess I would have to tear out the additions and revert to the official firmware.

But hopefully  the errors could be remedied soon, I would be glad to help troubleshoot .

Furthermore, the frozen screen doesn't even accept connections to my octopi, so the serial is also not working.

thisiskeithb, 2020-07-17:
That’s odd that the precompiled binary works and updating from bugfix/source does not. Those fixes were integrated here about two hours ago, so there shouldn’t be any difference.

mjmurray68, 2020-07-17:
It's likely something with my process then. let me go one step further and nuke everything I have local via GHDesktop and start completely fresh. I'll post back in a few minutes.

thisiskeithb, 2020-07-17:
Can you attach the top of your compile output where is shows all the library versions? Either attach it as a text file or use [pastebin](https://www.pastebin.com) would work.

fleek, 2020-07-17:
I need to check this logic.
`#if NO_EEPROM_SELECTED
  // FLASH
  //#define FLASH_EEPROM_EMULATION

  // I2C
  #define IIC_BL24CXX_EEPROM                      // EEPROM on I2C-0 used only for display settings
  #if ENABLED(IIC_BL24CXX_EEPROM)
    #define IIC_EEPROM_SDA                  PA11
    #define IIC_EEPROM_SCL                  PA12
    //#define MARLIN_EEPROM_SIZE 0x4000           // 16Kb (24c16)
  #endif

  #define SDCARD_EEPROM_EMULATION                 // SD EEPROM until all EEPROM is BL24CXX
  #define MARLIN_EEPROM_SIZE 0x1000               // 4Kb

  // SPI
  //#define SPI_EEPROM                            // EEPROM on SPI-0
  //#define SPI_CHAN_EEPROM1  ?
  //#define SPI_EEPROM1_CS    ?

  // 2K EEPROM
  //#define SPI_EEPROM2_CS    ?

  // 32Mb FLASH
  //#define SPI_FLASH_CS      ?
#endif`

Where do I need to put the "NO_EEPROM_SELECTED" flag?  It is neither in Configuration.h nor Configuration_adv.h

fleek, 2020-07-17:
> Can you attach the top of your compile output where is shows all the library versions? Either attach it as a text file or use [pastebin](https://www.pastebin.com) would work.

`PACKAGES:
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerialM> 1.0.0
|-- <Adafruit MAX31865 library> 1.1.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SlowSoftI2CMaster> 1.0.0
|-- <LiquidTWI2> 1.2.7
|   |-- <Wire> 1.0
|-- <SailfishLCD>
|-- <SoftwareSerialM> 1.0.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <STM32ADC> 1.0
|-- <Wire> 1.0`

thisiskeithb, 2020-07-17:
> Where do I need to put the "NO_EEPROM_SELECTED" flag?

You don't. `NO_EEPROM_SELECTED` is a macro:

https://github.com/MarlinFirmware/Marlin/blob/c2363a81b47a4c5ced63fa0765c67c0a4ccf462f/Marlin/src/inc/Conditionals_adv.h#L366-L369

It'll default to SD card emulation per the `CREALITY_V4` pins file.

fleek, 2020-07-17:
> Can you attach the top of your compile output where is shows all the library versions? Either attach it as a text file or use [pastebin](https://www.pastebin.com) would work.

[dependencies.txt](https://github.com/MarlinFirmware/Marlin/files/4939954/dependencies.txt)

thisiskeithb, 2020-07-17:
@mjmurray68: I just realized I wasn't replying to you. Please attach the top part of your log output as well.

For good measure, I've compiled the default Ender-3 V2 config firmware on my system: [ender-3-v2-7-17-2020.zip](https://github.com/MarlinFirmware/Marlin/files/4939980/ender-3-v2-7-17-2020.zip)

@fleek: Those are just warnings and can be ignored.

mjmurray68, 2020-07-17:
@thisiskeithb after re-building all of my repository forks, clones and branches, I was able to successfully build and flash the generic bugfix-2.0.x code. 

For good measure here is the compile header:
Processing STM32F103RET6_creality (platform: ststm32@~6.1.0; board: genericSTM32F103RC; framework: arduino)
-------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.1.1 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (jlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerialM> 1.0.0
|-- <Adafruit MAX31865 library> 1.1.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SlowSoftI2CMaster> 1.0.0
|-- <LiquidTWI2> 1.2.7
|   |-- <Wire> 1.0
|-- <SailfishLCD>
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2

thisiskeithb, 2020-07-17:
So does your Ender-3 V2 behave as expected?

mjmurray68, 2020-07-17:
It does. I'm all set to close, just didn't want to cut off fleek if they were looking for more information.

fleek, 2020-07-17:
@mjmurray68 you can close this, sorry for highjacking your thread.  I will open up a new issue if I encounter any problems.

mjmurray68, 2020-07-17:
Combination of early release, and timing issues with commits.
Thanks to everyone who helped me figure this out!

closed 2020-07-17 by mjmurray68

thinkyhead, 2020-07-20:
When Chuck Hellebuyck had this issue, he said he solved it by just renaming the `firmware.bin` with a unique name. You have to do that with this style of board, because it ignores a file that has the same name as the last-flashed file.

thisiskeithb, 2020-07-20:
That’s....different, but good to know.

thinkyhead, 2020-07-20:
I've added a README to the example configurations. That should eliminate all support requests for this issue.