MarlinFirmware/Marlin #18676: [BUG] Bugfix 2.0.x Breaks Neopixel on SKR mini E3 DIP

dotdash32, 2020-07-16:
### Bug Description

Updated to bugfix-2.0.x, and now Neopixels do not light up.  Yesterday, was using a build from 2.0.x release from maybe a month ago, and had to do some reworking to get the STM32F1 to compile, but it did eventually work.  Updated to bugfix to fix those issues and use neopixel indexing, and pixels do not light.  I'm wiring through the buzzer pin, and yesterday could hear slight buzzing as scrolled through led settings, but not now.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4934916/config.zip)
Whole branch is on my Dotdash branch on my [Marlin fork](https://github.com/dotdash32/Marlin/tree/Dotdash/Marlin).

### Steps to Reproduce

1. Updated to bugfix 2.0.x
2. Updated Neopixel library version in `platformio.ini` to `1.2.4` per [this RepRap post](https://reprap.org/forum/read.php?415,874307).  This had been previously done before updating to bugfix as well.
3. Remove Adafruit Neopixel from lib-ignore, leaving only SPI.
4. Compile w/ `STM32F301RE_btt_USB`

**Expected behavior:** 

Neopixels light up with `STARTUP_TEST` and can be controlled.  Alternatively, if something is wrong with the pixels, since the signal is going through the buzzer pin, a sound should be audible while sweeping a custom LED color set (ie adjusting from `0->255` on any component).

**Actual behavior:** 

No pixels light up, and sound is not audible.  I don't have an O-scope at home to see if the signal is coming, the buzzer test is my best bet.

#### Additional Information

Yesterday I figured out why I was having some packet loss (colors on neopixels didn't line up very well), likely due to the operating voltage mismatch of the STM32 and the pixels.  I installed some diodes on Vcc to drop the operating voltage and get the 3V3 signals above the threshold.  This worked well yesterday on my old build of Marlin.

Cable routing wise, the DIN pin goes through the LCD's buzzer, but I have not had any problems with this beforehand.

thisiskeithb, 2020-07-17:
Since Neopixels are part of the ignored libraries, this isn't a Marlin bug. You may have been able to get it work before, but it's disabled due to compatibility issues.

https://github.com/MarlinFirmware/Marlin/blob/11e29754e65353e5bff9a0a0d990d8fbd6c1bd07/platformio.ini#L341-L354

closed 2020-07-17 by thisiskeithb

dotdash32, 2020-07-17:
Makes sense, easiest fix would just be to use a different board with a support environment?

I'm using this for a computer vision system, any thoughts on similar systems to get at least some level of lighting without drawing directly from a GPIO pin?

thinkyhead, 2020-07-17:
If you do find some `Adafruit NeoPixel@1.?.?` that works, maybe we'll be able to include it for all.

dotdash32, 2020-07-17:
I did a bit more digging, and BigTreeTech has released an older version of the Neopixel library which does have the code for the STM32F1, so if you set the library to call `https://github.com/bigtreetech/Adafruit_NeoPixel/archive/master.zip`, it does work, but I'm pretty sure there are a couple bugs or at least my second pixel setup has some issues with skipping pulses.  I think it might be a mostly lost cause for the DIP, but I believe the 1.2 and 2.0 versions of the SKR mini E3 board have a level shifter so they can talk to pixels well.

I don't have the background to really understand the level of bit banging used for neopixel communication, but I think it might need to be actually integrated into Adafruit's main library before it can be easily integrated into Marlin's main body.