MarlinFirmware/Marlin #18677: [BUG] bugfix-2.0.x #define HOTEND_IDLE_TIMEOUT causing cold extrusion during prints and G26

GhostlyCrowd, 2020-07-17:
As the title states with it on, my hotend and bed will randomly decide to turn off mid print and even mid G26 validation.

Turned #define HOTEND_IDLE_TIMEOUT off and all is well again.

thisiskeithb, 2020-07-17:
Please follow the Issue Template when filing a bug report:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

GhostlyCrowd, 2020-07-17:
I'm a fool, built wrong pull, i'm behind. I'll do a pull and re-open if its still an issue.

closed 2020-07-17 by GhostlyCrowd

thisiskeithb, 2020-07-17:
There was a recent fix to Hotend Idle Timeout, so make sure you pull the latest `bugfix-2.0.x`.

GhostlyCrowd, 2020-07-17:
> There was a recent fix to Hotend Idle Timeout, so make sure you pull the latest `bugfix-2.0.x`.

Yep doing so now, thanks!