MarlinFirmware/Marlin #18681: [FR] Add support for MKS Robin R61505U-based TFT display

ashavakloo, 2020-07-17:
hello Everybody
I have a MKS Robin V2.3 board and Robin TFT V1.1R(with R61505U controller). I wanted to install Marlin-bugfix-2.0.x on it. but unfortunately R61505U-based TFT display is not supported by marlin.and i cannot write the needed init sequences.
i want to know is it a chance to add support for it? 
thanks in advance.

thisiskeithb, 2020-07-17:
@makerbase-mks: Any plans for this?

ashavakloo, 2020-07-17:
> @makerbase-mks: Any plans for this?

I contacted makerbase but they didn't help me

jmz52, 2020-07-17:
@ashavakloo
I've checked the datasheet for your display and it looks similar to ILI9325 controller. This is both good and bad.
Good part is that I've already added support for ILI9325 and might be able to adopt it for you display.
Bad part is that I have no relevant hardware and someone needs to to lots and lots of testing (setting correct screen orientation and color mode was not easy).
Are you willing to do the testing and do you know how to connect to COM port to read debug messages (plain test)?

ashavakloo, 2020-07-17:
thanks for replying jmz52
I changed Case 0x9328 to case 0x1505 In Marlin/scr/lcd/dogm/u8g_dev_tft_320x240_upscale.....cpp
The good news is the screen is working now but I have two problems:
1:The writings are reverse
2:TFT display is pale And it is difficult to see the writings in a bright place.
what should i do know?

> @ashavakloo
> I've checked the datasheet for your display and it looks similar to ILI9328 controller. This is both good and bad.
> Good part is that I've already added support for ILI9328 and might be able to adopt it for you display.
> Bad part is that I have no relevant hardware and someone needs to do lots and lots of testing (setting correct screen orientation and color mode for ILI9328 was not easy).
> Are you willing to do the testing and do you know how to connect to COM port to read debug messages (plain text)?

ashavakloo, 2020-07-19:
Can anyone help me?please😭

xC0000005, 2020-07-19:
Do you have a link to firmware that works with this? It might be a gamma correction setting. I can probably extract the init sequence used from that firmware.

ashavakloo, 2020-07-19:
https://github.com/makerbase-mks/MKS-Robin/blob/master/MKS%20Robin/MKS_ROBIN_V2.0.8%20Release%20file.rar
This link is related to the latest version of mks robin firmware (This is a closed source firmware).

> Do you have a link to firmware that works with this? It might be a gamma correction setting. I can probably extract the init sequence used from that firmware.

xC0000005, 2020-07-19:
That’s the configuration file. What firmware.bin would they have you install? I can decompile it and see how they’re intiialziing and if they’re doing custom gama correction/settings.

xC0000005, 2020-07-19:
I found the init sequence for this LCD but it’s a mess. We will see if it does anything special once I work out the complete path.

xC0000005, 2020-07-19:
I can't make much sense of this init sequence - the first thing they do is write to a register I'm not sure exists (and this is explicitly in this LCD's execution path)

[image]

ashavakloo, 2020-07-20:
> I can't make much sense of this init sequence - the first thing they do is write to a register I'm not sure exists (and this is explicitly in this LCD's execution path)
> 
> [image]

so there is no way to fix this?
can i replace this controller with ILI9328?

xC0000005, 2020-07-20:
We can certainly figure it out. It just may take time.

jmz52, 2020-07-20:
@ashavakloo 
Please update you MKS Robin with firmware linked below and check debug information on COM port (250000 connection speed)
https://github.com/jmz52/bin/blob/master/R61505U/Robin.bin

ashavakloo, 2020-07-20:
@jmz52
I updated my MKS Robin with your firmware.
[image]
[image]

>  check debug information on COM port (250000 connection speed)

I don't know how to check debug information.
It is possible to explain a bit please?

xC0000005, 2020-07-20:
This code:
https://github.com/nopnop2002/wiringpi-tft-tool/blob/6f0f91fed2d5a07f1e62663eb8bc37458c11bb65/8bitlib.c <https://github.com/nopnop2002/wiringpi-tft-tool/blob/6f0f91fed2d5a07f1e62663eb8bc37458c11bb65/8bitlib.c> 

Is structured very similarly to the code in the MKS robin binary, and it also includes a write to the 0xE5 register for this LCD. In fact, the code is so similar I’m wondering if it’s a common init sequence provided by the manufacturer, or someone copied from someone else.

jmz52, 2020-07-20:
You need to connect USB to your computer and use terminal program (PuTTY for example) to view information send by Marlin.
Alternatively you can use Repetier Host program - it shows logs from printer in lower part of the window.

ashavakloo, 2020-07-20:
>check debug information on COM port (250000 connection speed)

[image]
what should i do now?

jmz52, 2020-07-21:
@ashavakloo 
I've stripped init sequence from non-existing regusters, please re-test [Robin.bin](https://github.com/jmz52/bin/blob/master/R61505U/Robin.bin) and post photo to make sure that image is unchanged.

Also please test 5 more variants in https://github.com/jmz52/bin/tree/master/R61505U and post photos.
These variants have slightly different init sequences, I need to figure out what various parameters are responsible for.

ashavakloo, 2020-07-21:
>please re-test [Robin.bin](https://github.com/jmz52/bin/blob/master/R61505U/Robin.bin) and post photo to make sure that image is unchanged.

it didn't change
[image]
[image]

> Also please test 5 more variants in https://github.com/jmz52/bin/tree/master/R61505U and post photos.

variant0-ili9328: writings are mirrored
[image]
[image]
variant1:lcd worked well but its inverted
[image]
[image]
variant2:just like variant0-ili9328
[image]
[image]
variant3:like variant1 in diffrent coller but it's not inverted
[image]
[image]
variant4:bootscreen is inverted and wrintigs are mirrored
[image]
[image]

jmz52, 2020-07-21:
Please test new init sequences - variants 5-9

ashavakloo, 2020-07-21:
> Please test new init sequences - variants 5 - 10

variants 5 :LCD is inverted,writings are mirrored and in info screen touch has some problems for example when i touched fan, nozzle 1 setting were displayed and when i touched nozzle 2 fan setting were displayed.
[image]
[image]
variants 6 :everything works well and touch is like variants 5
[image]
[image]
variants 7 :LCD is inverted,writings are mirrored and touch has many problems
[image]
[image]
variants 8 :writings are mirrored and touch is like variants 5
[image]
[image]
variants 9 : :like variants 5 
[image]
[image]

xC0000005, 2020-07-21:
If you run variant 6, and when you get to the status screen, press and hold in the upper left corner until calibration starts, once it calibrates, does touch work as expected? (I have to use a stylus to calibrate, but not after that).

jmz52, 2020-07-22:
@ashavakloo
Have you tried variant 10?

Variant #6 does not work well - Y positioning is inverted
Here is how screen should look like
[image]

ashavakloo, 2020-07-22:
> once it calibrates, does touch work as expected?

unfortunately no

> Have you tried variant 10?

I've tested it and it works well 
[image]

jmz52, 2020-07-22:
Hooray! 
Now can you please connect PuTTY and calibrate your screen?
Calibration data will be printed to COM port once screen is successfully calibrated.

ashavakloo, 2020-07-22:
> Now can you please connect PuTTY and calibrate your screen?

Touch screen calibration completed
TOUCH_CALIBRATION_X 12489
                         TOUCH_CALIBRATION_Y 9210
                                                 TOUCH_OFFSET_X -52
                                                                   TOUCH_OFFSET_                                                                                                             Y -17
     TOUCH_ORIENTATION TOUCH_LANDSCAPE

jmz52, 2020-07-23:
@ashavakloo 
New code, configuration files and build environment are available at https://github.com/jmz52/Marlin/tree/R61502
Make sure you set right calibration data in Configuration.h in TOUCH_SCREEN section.

ashavakloo, 2020-07-24:
@jmz52 
thank you for your help 
I still have a problem
When want to compile the your firmware, I get g ++ not found error,

jmz52, 2020-07-25:
Is this issue specific for modified code or you have the same problem when compiling Marlin from official repository?

ashavakloo, 2020-07-25:
> Is this issue specific for modified code or you have the same problem when compiling Marlin from official repository?

no,I think yesterday i did something wrong
now i can compile and use this firmware without any problem.
Thank you very much

ashavakloo, 2020-07-25:
thank you 
@jmz52 
@thisiskeithb 
@xC0000005 
for help 

should I close this issue?

ashavakloo, 2020-07-31:
i know my problem is Not related to this issue 
when i want to do manual mesh bed leveling, after Homing XYZ ,Click to Begin appears.
but nothing happens when i tap on LCD
[image]
what should i do?

ashavakloo, 2020-08-01:
> i know my problem is Not related to this issue
> when i want to do manual mesh bed leveling, after Homing XYZ ,Click to Begin appears.
> but nothing happens when i tap on LCD
> [image]
> what should i do?

@jmz52 please help me.
I will be grateful if you can help me.

jmz52, 2020-08-01:
@ashavakloo
I've never used mesh bed leveling, so I need your help to reproduce this problem.
Please provide your configuration files and name of menu item that gets you to this screen.

ashavakloo, 2020-08-01:
First of all, I have to say thank you for your help
> Please provide your configuration files 

[Configuration&Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5011048/Configuration.Configuration_adv.zip)

> name of menu item that gets you to this screen.

I do not understand exactly what you mean but mesh bed leveling settings are between lines 1327 to 1456
in LCDs like RepRapDiscount Smart Controller, mesh bed leveling begin when you press the encoder

jmz52, 2020-08-01:
Which menu item you selected (when navigated Marlin's menu) to get to this "Click to Begin" screen?

ashavakloo, 2020-08-02:
> Which menu item you selected (when navigated Marlin's menu) to get to this "Click to Begin" screen?

Menu/motion/bed leveling/level bed

sjasonsmith, 2020-08-17:
Is this working now? I'm not sure whether this is related to all the work @rhapsodyv has done recently to improve TFT support.

rhapsodyv, 2020-08-17:
> Is this working now? I'm not sure whether this is related to all the work @rhapsodyv has done recently to improve TFT support.

I ported new color ui to STM32F1. If the user problem is related with STM32, he can compile his board for STM32F1 and give try.

@ashavakloo can you try the last bugfix and target stm32f1?

sjasonsmith, 2020-08-17:
The only configs I saw were an MKS Robin, which is already STM32F1. A couple board (maybe Robin Pro and Robin2) are STM32F4.

rhapsodyv, 2020-09-03:
The new upcoming TFT code will allow set the driver, rotation and mirroring. So you may be able to configure your display without any problem.

jmz52, 2020-09-07:
@ashavakloo, I am back. Will be able to dig into this in a day or two.
My guess - this problem is caused by some part of classic UI code that needs to tweaked to work with touch screen and no physical encoder.
This should not be a big problem to fix - there were number of such places, and the most troublesome was UBL support with it's direct interfacing with rotary encoder.

ashavakloo, 2020-09-21:
@jmz52
sorry I haven't been for a while 
you said you Will be able to dig into this in a day or two.
How can I help you to solve this problem?
Stay healthy.

jmz52, 2020-09-21:
@ashavakloo 
I've found the root cause of this problem. Unfortunately it is an oversight in touch UI design (caused by lack of knowledge of some menu items' behavior).
I have an idea how to fix it, but I need more time to do it.

jmz52, 2020-09-27:
@ashavakloo, I've updated UI code to emulate clicks on TFT screens for bed leveling
You can download it from https://github.com/jmz52/Marlin/tree/tft_and_touch_N2

ashavakloo, 2020-10-24:
@jmz52
Thank you so much 

> You can download it from https://github.com/jmz52/Marlin/tree/tft_and_touch_N2

I used it but unfortunately Touch does not work at all 

[Configuration&Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5432955/Configuration.Configuration_adv.zip)