MarlinFirmware/Marlin #18690: [BUG] G29 Not centered on the bed

temakitua, 2020-07-18:
Hello,

I have some troubles with the g29 feature.
It looks like the feature doesn't get the bed size and the probe offset (from extruder) correctly.
 
[Marlin-bugfix-2.0.x (2).zip](https://github.com/MarlinFirmware/Marlin/files/4941373/Marlin-bugfix-2.0.x.2.zip)

Here a liltre shematic to show you the situation :
[image]

Thanks guys for you help.
Already tried to set it up with but it allow me to recentered g29 but took too much margin above the edge :

#if PROBE_SELECTED && !IS_KINEMATIC
  #define PROBING_MARGIN_LEFT (46 + PROBING_MARGIN)
  #define PROBING_MARGIN_RIGHT (PROBING_MARGIN)
  #define PROBING_MARGIN_FRONT (1 + PROBING_MARGIN)
  #define PROBING_MARGIN_BACK (PROBING_MARGIN)
#endif

thisiskeithb, 2020-07-18:
Does your nozzle home off the bed? You have both `X_MIN_POS` and `Y_MIN_POS` set to `0` in your config.

cr20-123, 2020-07-18:
I had the same problem as you, to fix it I had to increase the probing_margin. Try Increasing it to 50 and see if then probes the center of the bed. I've noticed that you can run out of x travel since the probe in your case is 46mm to the left and before you can get the probe to the edges of the bed you reach your x bed size. To accommodate for that it looks like marlin shrinks the probing area to where it thinks it will be able to probe. By increasing the probing_margin it should make a smaller centered box that can probe the center. I made the changes to your firmware and attached it below. Hope this helps.

[Marlin-bugfix-2.0.x new probe margin.zip](https://github.com/MarlinFirmware/Marlin/files/4942143/Marlin-bugfix-2.0.x.new.probe.margin.zip)

temakitua, 2020-07-18:
> Does your nozzle home off the bed? You have both `X_MIN_POS` and `Y_MIN_POS` set to `0` in your config.

Hey, when I am homing, my probe is centered. but the nozzle is not centered (X,Y offset) that is normal for me.

temakitua, 2020-07-18:
> I had the same problem as you, to fix it I had to increase the probing_margin. Try Increasing it to 50 and see if then probes the center of the bed. I've noticed that you can run out of x travel [...]

Yes I discovered the same thing, but it doesn't really fix the problem because now the g29 mesh is centered but don"t calculate on the whole bed. 
May be should try to modify the x_min_pos as suggered by @cr20-123

cr20-123, 2020-07-18:
> Yes I discovered the same thing, but it doesn't really fix the problem because now the g29 mesh is centered but don"t calculate on the whole bed.
> May be should try to modify the x_min_pos as [...]

Try uncommenting EXTRAPOLATE_BEYOND_GRID

closed 2020-07-18 by thisiskeithb