MarlinFirmware/Marlin #18691: [BUG] (Print immediately purges and resumes after pausing)

Liamnerf, 2020-07-18:
### Bug Description

Upon pausing a print, the nozzle parks, immediately begins to purge itself, then resumes the print without prompting. Running marlin bug fix

### My Configurations
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4941735/Configs.zip)

### Steps to Reproduce

1. select "pause print" from the LCD menu

**Expected behavior:** 

Upon pausing, the nozzle should park itself and remain in place, waiting for a click of the LCD knob before restarting the print.

**Actual behavior:** 

The nozzle parks, immediately begins to purge itself, then resumes the print without prompting.

qwewer0, 2020-07-18:
https://github.com/MarlinFirmware/Marlin/issues/18191 related?

Liamnerf, 2020-07-18:
> #18191 related?

Possibly, though what's being described there doesn't exactly fit with what I'm experiencing.

Liamnerf, 2020-07-19:
Closing this now, solved it by updating to Marlin 2.0.5.4.

closed 2020-07-19 by Liamnerf