MarlinFirmware/Marlin #18693: [BUG] Encoder wheel can miss step when reversing direction

marknn3, 2020-07-18:
### Bug Description

When reversing the encoder wheel direction, the first step can be missed.
A user will perceive this as unreliable input: a rotation click without any update.

### Configurations

This affects all configurations with ENCODER_PULSES_PER_STEP > 1

### Steps to Reproduce

Move the encoder wheel fast or slow in one direction. Stop and reverse direction by 1 click.
The first reverse click is randomly ignored.

qwewer0, 2020-07-18:
I might have this too, on Ender 3 12864 display, or having an unreliable rotary encoder.

GMagician, 2020-07-18:
I saw it but never considered a bug, but yes it's not normal. I'll test your PR

thinkyhead, 2020-07-18:
I see that `encoderDiff = 0` only happens when `encoderPastThreshold`. See if it helps to move this line down after the condition. This would be line 950 in the current code. That will prevent it from having any residual value in-between screen updates.

boelle, 2020-07-27:
@marknn3 i see you PR is merged, did that help?

GMagician, 2020-07-27:
From some test I did it seems to solve issue

sjasonsmith, 2020-07-28:
@marknn3 based on testing from @GMagician I'm closing this issue, with the assumption that your merged change fixes it.

If that assumption ends up being correct, please let us know.

PR #18694 (2020-07-19): Reliable encoder wheel action when reversing direction
### Description

This change will fix a reliability issue with the encoder wheel when reversing direction.
After a direction change the first step was randomly ignored.
This was caused by missed encoder inputs. In most situations this is unnoticed. But when reversing direction, this can lead to not registering the first step.

By detecting a direction change under strict conditions, this behavior has been corrected.

This fixes #18693

### Benefits

The user will have reliable encoder wheel action.
My personal reaction was that I thought I [... cut]
PR #18754 (2020-07-24): Improve and simplify encoder wheel handling
### Description
This change will properly 'timeout' the residual encoder step value, and therefor obsoletes the complex 'reversal' check.

### Benefits
It simplifies the code and actually improves the behavior of the encoder wheel handling.

### Related Issues
This fixes #18693
closed 2020-07-28 by sjasonsmith