MarlinFirmware/Marlin #18704: [BUG] Error compile btt skr mini v1.1

userosos, 2020-07-19:
Hi! I download latest marlin firmware - Marlin-2.0.5.4. I use Visual Studio Code and Platformio. I use env:STM32F103RC_btt_usb. I cant make the firmware because i has the error.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4943603/configs.zip)
Logs:
https://pastebin.com/6sCZwV83

ellensp, 2020-07-19:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

But if you actually read all of this, see issue 2 in https://reprap.org/forum/read.php?415,874307

closed 2020-07-19 by ellensp

userosos, 2020-07-19:
@ellensp 
Great! Thx! 
I find the line "Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip" and replace it with
"Adafruit MAX31865 library@~1.1.0"

In every release i have some errors with make marlin (i use different motherboards). In older versions i has no problem but now i has some errors all time. Write to me Is this a new reality or will everything be ok over time?
Big thx for the support!

ellensp, 2020-07-19:
marlin used to request latest libraries, but latest libraries started breaking marlin. Future releases will specify particular version of library it is known to work with that version of marlin.

userosos, 2020-07-19:
> marlin used to request latest libraries, but latest libraries started breaking marlin. Future releases will specify particular version of library it is known to work with that version of marlin.

Good! I hope there will be fewer problems like this with every release! Otherwise, you can think about creating an oldstable branch for people like me. Who needs 100% working version. Thanks again for your help and a good product!