MarlinFirmware/Marlin #18706: [FR] (feature request title)

baera, 2020-07-19:
### Description

Re-Organisation of the configuration*.h files

### Feature Workflow

I have spent days working on my LPC1769 based board, going through all the options, compiling and testing time and time again. Yes, Marlin is an amazing piece of work, especially considering it is done mostly by volunteers.

The configuration is currently a one-size-fits-all. If it is a primitive 2.5D printer or the most advanced multi-hotend device with splicer and 3 Z-axis motors.

But: Most users are looking for a simple way to configure just those features they have on hand, compile, upload and test.
Many are experiencing different results between the current branch and the bugfix one (latter won‘t compile for LPC1769 in my case).

1. Coffee
2. Define use-cases (most are upgrades of existing printers due to a different hot-end, BlTouch, etc.)
3. Split the configuration or make it pre-configurable: Perhaps a simple python script asking what kind of setup it is and spitting out a simplified version of the configuration file

#### Additional Information

I am used to working with ESP32 microcontrollers, armhf and arm64 embedded systems. Most if them come with a cli with a menu that makes configuring „easier“.
Using pio may be neat, but under OSx it is useless in my case (compile once, have to clean up after, then compile, or it won‘t work at all - and it is a pio bug, not a Marlin of config error).

What could I contribute? I would be prepared to serve and support a website, where successful builds and their configuration can be shared.
From the configuration files and printer type one could distill prepared configuration files for future users.
The configuration files could also be much leaner just by removing all the if x enabled then y.
Perhaps viewing each _option_ like calling a function with parameters would be sensible.

Tbh that‘s what makes Kay3D special, but not perfect.

Maybe it‘s the complexity that’s getting out of hand slowly - while users are crashing their nozzles in to build platforms. 

This is no critique of the fabulous work done - I just think one needs to sit back, clear the screen and look at it from a users perspective. A poll my be a good idea too.

I‘m here to help.

ellensp, 2020-07-19:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-19 by ellensp

baera, 2020-07-19:
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

I take this as a „no“. 

Reminds me of this: 
[image]

baera, 2020-07-19:
[image]