MarlinFirmware/Marlin #18707: Mesh bed doesn't work with dual-x. I have this question for a long time. Could you help? [BUG] (short description)

Ziyayildizz3, 2020-07-19:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Ziyayildizz3, 2020-07-19:
H

boelle, 2020-07-19:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

AnHardt, 2020-07-19:
There is absolutely no way haw this could work.
If one nozzle is in a valley and the other on a hill - what should the system do? Ram one nozzle into the hill?

Ziyayildizz3, 2020-07-19:
> Bu Sayı Kuyruğu, Marlin hata raporları ve geliştirmeyle ilgili sorunlar içindir ve burada kullanıcı destekle ilgili soruları ele almamayı tercih ediyoruz. ( [Bu sayfada](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question) belirtildiği gibi .) Yapılandırma ve sorun giderme konusunda yardım almak için lütfen aşağıdaki kaynakları kullanın:
> 
> * MarlinFW.org [Marlin Belgeleri](http://marlinfw.org)
> * RepRap.org [Marlin Forumu](http://forums.reprap.org/list.php?415)
> * [Tom'un 3D Forumları](https://discuss.toms3d.org/)
> * Facebook Grubu ["Marlin Yazılım"](https://www.facebook.com/groups/1049718498464482/)
> * Facebook Grubu ["3D Yazıcılar için Marlin Yazılım"](https://www.facebook.com/groups/3Dtechtalk/)
> * [](https://www.youtube.com/results?search_query=marlin+configuration)YouTube'da [Marlin Yapılandırması](https://www.youtube.com/results?search_query=marlin+configuration)
> * Marlin Discord sunucusu. Üyelik bağlantısı: https://discord.gg/n5NJ59y
> 
> Topluluktan yardım istedikten sonra, fikir birliği Marlin'de bir hataya işaret ediyorsa, bir [hata raporu](https://github.com/MarlinFirmware/Marlin/issues/new/choose) göndermelisiniz .

This marlin error. With Marlin 1.1.6, mesh-Bed works smoothly. When I say table level in the new version, the axes go home and do nothing. I have been using the prayer-x system for a long time.

Ziyayildizz3, 2020-07-19:
> Bunun nasıl çalışabileceğinin kesinlikle bir yolu yok.
> Bir meme bir vadide, diğeri bir tepede ise - sistem ne yapmalıdır? Tepeye bir meme mi koydun?

There are two independent x-axes. The two nozzle levels are the same. After X1 prints, it parks and X2 exits. Works seamlessly with Marlin 1.1.6 Mesh-bed.

boelle, 2020-07-20:
dont use marlin 1.1.6 use [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)

qwewer0, 2020-07-20:
@boelle The configuration files that @Ziyayildizz3 provided shows that the problem is with the Bugfix-2.0.x, but it worked/works fine on Marlin 1.1.6.

> CONFIGURATION_H_VERSION 020006
CONFIGURATION_ADV_H_VERSION 020006

Ziyayildizz3, 2020-07-20:
@qwewer0 Absolutely. This problem has existed for a long time. I have forwarded several times but the error raparu has been removed.

thinkyhead, 2020-07-20:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

thinkyhead, 2020-07-20:
In addition to the above, please move the nozzle to some spots around the bed and send `M114 D` to get a report of how leveling is being applied at that position. That should give some idea how Marlin is calculating offsets at different positions.

thinkyhead, 2020-07-20:
And a note to Triage people: Pay attention to the issue and don't just post automatic responses. I'm a bit embarrassed how poorly this issue has been addressed. Watch me carefully for the right way to follow up on issues.

thinkyhead, 2020-07-20:
While I'm waiting for Ziya to come back with some logs, I will be looking through the mesh leveling code to see if anything obvious jumps out at me, and to see where I might want to add logging or ask new questions to fill in our knowledge gaps.

thinkyhead, 2020-07-20:
One of the steps needs more detail for `MESH_BED_LEVELING`…

- Do a **`G29 S1` followed by `G29 S2`** to probe the bed. This will also enable bed leveling.

Ziyayildizz3, 2020-07-20:
@thinkyhead Thankyou. I tried to shoot the problem with the video. I will try what you say and give feedback.
https://www.dosya.tc/server29/eds4yg/IMG_3357.zip.html

thinkyhead, 2020-07-20:
I don't need video, thanks. I believe the problem exists.

Ziyayildizz3, 2020-07-20:
Thank you for your kind reply and interest. I will collect the data following the instructions and forward them to you.

InsanityAutomation, 2020-07-22:
Ive got an idex machine with a probe, I can run a test on it this weekend if nothing comes back before then.

closed 2020-08-27 by github-actions[bot]