MarlinFirmware/Marlin #18712: [BUG] PRUSA_MMU2_S_MODE reboots on Tx/C0 command timeout

isanval, 2020-07-19:
### Bug Description

PRUSA_MMU2_S_MODE do not retries correctly when Tx/C0 commands timeouts from MMU2. Instead retry, it reboots Marlin.

Marlin SKR1.3 + BLTouch + MMU2S

### My Configurations

[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/4944376/MarlinConfig.zip)

### Steps to Reproduce

1. Upload included firmware
2. Send T0 command to Marlin without letting MMU2 to successfully load the filament.

**Expected behavior:**
Marlin retries the T0 command 5 times until it gives up.

**Actual behavior:**
Marlin reboots after the first timeout

thisiskeithb, 2020-07-19:
Is this the same issue you commented about in your "PRUSA_MMU2_S_MODE enhacement, spin the BMG gears while C0 command" feature request? https://github.com/MarlinFirmware/Marlin/issues/18599

isanval, 2020-07-20:
Yes, I think so. The point is that the bug I was facing in the FR is not mine but inherited from the original PRUSA_MMU2_S_MODE development.

Anyway, the FR is still valid, because although the bug is there, in the original PRUSA_MMU2_S_MODE the E0 motor is not spinning, so it is still valid.

isanval, 2020-07-22:
I think I found the bug. The function "mmu2_not_responding()" is being called by "manage_response function()" when we get the first timeout:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/mmu2/mmu2.cpp#L813

The function is declared as follows:
```
static bool mmu2_not_responding() {
  LCD_MESSAGEPGM(MSG_MMU2_NOT_RESPONDING);
  BUZZ(100, 659);
  BUZZ(200, 698);
  BUZZ(100, 659);
  BUZZ(300, 440);
  BUZZ(100, 659);
}
```

But it's not returning any value, no "return" call at all. Placing a "return(true)" at the end of the function, Marlin is not hanging up any more.

Not sure what should return or if it should be declared as "static void mmu2_not_responding()", but this is what is causing the reboot in Marlin.

Hope this can helps to fix that in the proper way.