MarlinFirmware/Marlin #18713: [BUG] Filament change in 3D printer with DGUS (DWIN Graphical Utility Software) LCD - no "Extrude more" option

jopl, 2020-07-19:
### Bug Description

Dear Marlin firmware developers,
I have got troubles with my 3d printer with FYSETC S6 mainboard onboard connected to DGUS (DWIN Graphical Utility Software) LCD. Everything works like a charm, but filament change is not optimal. I would like to have option to extrude more filament after filament change, but unfotunately there is just possibility to terminate the change process. Is there any solution to solve such issue? Would you be so kind and could you advise me also with the screen (number and name), which should be optimal to serve such event, please?
Thank you very much in advance.
Kind regards, jopl.

### My Configurations

FYSETC S6
DGUS (DWIN Graphical Utility Software) LCD

Config files are in zip attached

### Steps to Reproduce

1. Start print
2. Filament change (M600 command via serial port)

**Expected behavior:** [What you expect to happen]

Marlin should wait for
a) Filament change process termination confirmation and continue with printing
or
b) Extrude more filament choice to have clean material on output of extruder

**Actual behavior:** [What actually happens]

Marlin FW waits just for termination of filament change process and continues with printing

[Marlin_FYSETC_DGUS_cfgs.zip](https://github.com/MarlinFirmware/Marlin/files/4944397/Marlin_FYSETC_DGUS_cfgs.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-07-19:
It doesn't seem to be a bug, but more of a missing feature that a [Feature Request] would cover.

fleek, 2020-07-28:
This is also a problem for Ender 3V2 with DWIN DGUS controller.  Basically it beeps but no message is being shown on the the screen because the English message file does not contain the messages.  Will need further investigation on how to update the .bin file to upload to the DWIN DGUS controller.