MarlinFirmware/Marlin #18714: [FR] Multi Language Support for Host Prompts

mehmetsutas, 2020-07-19:
### Description

For most of the host prompts the wording is included in the C code and for a few of the prompts wording is linked to the language files. e.g. MSG_REHEATING, MSG_NOZZLE_PARKED, MSG_HEATER_TIMEOUT, MSG_PRINT_DONE.

Both ways are acceptable:

* The wording may be included in the C code and the host may be expected to present the prompts to the user and make any translation before presenting it to the user

or

* The wording may be linked to the language file and host just presents the prompts to the user without any translation.

My personal choice would be to link the wording to the language file so that each host software does not need to track the changes in Marlin, does not need to keep a list of the prompts and their translations and just presents the prompts to the user.

However linking some of the wordings to language files and including some in the C code is not acceptable. It makes it impossible to handle the prompts on the host side. Which one will be translated? If not translated, some prompts will be presented in English and some in local language.

We may even call the current situation a bug.

### Feature Workflow

My preferred solution is to link all the prompt messages including the Continue_Str and the Dismiss_Str to language files.

#### Additional Information

I just raised this issue in order to discuss and decide on the subject. When decided I can create a pull request for the agreed solution.

mehmetsutas, 2020-07-20:
@InsanityAutomation @thinkyhead @foosel @repetier

Your opinions would be highly appreciated.

repetier, 2020-07-21:
It is not acceptable to think all hosts would do translations. This is an illusion with all the firmwares and forks and ongoing changes the host writers have no real chance to know all used terms and offer translations. So the way to go is that firmware offers all terms in it‘s selected language which should be one the user understands. These are messages from the firmware so all hosts are expected to do is show them.

foosel, 2020-07-21:
I with @repetier on this one.

InsanityAutomation, 2020-07-21:
Agreed this can and should be handled in Marlin and host just faithfully displays it. Host involvement is to sit back and drink beer :-P

foosel, 2020-07-21:
I'm pretty sure the octopus is as adverse to alcoholic drinks as I am, so we'll stick to soft drinks if that's ok :-P

mehmetsutas, 2022-03-07:
This should be handled at the printer controller side. (This was argued at the Marlin repository.) Otherwise this would create a problem on many hosts whenever someone changes a host prompt or adds a new host prompt on Marlin. Also there are many printer controller firmware how will the hosts handle and translate each firmwares non standard prompts.

thinkyhead, 2022-04-21:
We are now doing translated host prompts, so if any have been missed please submit patches as you find them.

closed 2022-04-21 by thinkyhead