MarlinFirmware/Marlin #18715: [BUG] There is a problem with the Dual-X Mesh bed.

Ziyayildizz3, 2020-07-19:
Mesh bed is working with Dual-x in Marlin 1.1.6. However, in version 2.0, dual-x with mesh-bed does not work. Axes are reset and returned to the home screen with the level command. I ran mesh-bed with different version. I created the slope g-code with M503. I introduced the marlin version 2.0 again and introduced the slope with g-code. But mesh-bed is not working again !!!

thisiskeithb, 2020-07-19:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/18707

thisiskeithb, 2020-07-19:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-19 by thisiskeithb

thinkyhead, 2020-07-20:
@thisiskeithb — You don't think this should be investigated as a bug with `MESH_BED_LEVELING` in combination with `DUAL_X_CARRIAGE`?

thisiskeithb, 2020-07-20:
[This comment](https://github.com/MarlinFirmware/Marlin/issues/18707#issuecomment-660700264) from this original issue made it sound like an invalid combination.

AnHardt, 2020-07-20:
My bad. (Issue initially without configs - i was guessing)
For any mode where more than one nozzle is on the bed at the same time, the best bed leveling you can hope for to work, is 'linear', but not the mesh based.
In general this is true for `DUAL_X_CARRIAGE` but there is an exception i have not had on my screen - `AUTO_PARK`, where only one nozzle is on the bed at a time.

thisiskeithb, 2020-07-20:
I've reopened https://github.com/MarlinFirmware/Marlin/issues/18707.

Ziyayildizz3, 2020-07-20:
@AnHardt Yes I am using "AUTO_PARK_MODE". But mesh-bed does not work again. Marlin 1.1.6'dan beri çok gelişti. Yeni özellikleri seviyorum. Ancak ne yazık ki, yeni sürümü kullanamamamın nedeni, Dual-x'in örgü yatak seviyelendirme ile çalışmaması ...

thinkyhead, 2020-07-20:
Thanks for re-opening. Dual-X is not always "Dualing."