MarlinFirmware/Marlin #18716: Auto home with Skr 1.4 Turbo Tmc 2209 doesn't work have tried all the published fixes. Just goes 1cm and bounces

wyzarddoc, 2020-07-19:
Loaded Marlin 2.0 into my SKR 1.4 Turbo with TMC 2209 drives and all it does is move toward home bounces and stops. I have the TMC 2209's configured with software endstops in uart mode. I can move the axis near the stop by exceeding the bed with then homing works.  Anytime I try to home it bounces then resets the screen to 2mm which is the offset I programmed. I think either their is a bug in the module for the 2209 that senses the start-up current and thinks that's a overcurrent endstop trigger, or the timeout for autohome is set too low.

thisiskeithb, 2020-07-19:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-19 by thisiskeithb

sjasonsmith, 2020-07-20:
@wyzarddoc, in addition to going to support forums, please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. There were some dependency updates that impacted sensorless homing for some people.

wyzarddoc, 2020-07-20:
When you have tried all of the available solutions to a problem it is a bug not a support issue. Marlin is so bloated that it is starting to fail and will only get worse in my 40+ years of programming experience

thanks

thinkyhead, 2020-07-20:
I am applying my 43 years of coding experience every day towards improving the code in spite of its constant growth, and I don't entirely concur with your assessment.