MarlinFirmware/Marlin #18720: [BUG] After incorporation of (#18699) into bugfix compilation fails

bill-orange, 2020-07-20:
Bug Description: failure to compile after change described as "Only download & compile required libraries (#18699)"  in VSCODE

My Configurations: attached  
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4945413/Marlin.zip)

Steps to Reproduce:

1. Attempt to compile in the default manner
2. Observe errors.

Expected behavior: Successful compilation

Actual behavior:  Failure to compile in the usual default method

Additional Information:

There is more but I suspect that only the first error is relevant.

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CC -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-features-dependencies.h
'CC' is not recognized as an internal or external command,
operable program or batch file.
CalledProcessError: Command 'CC -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-features-dependencies.h' returned non-zero exit status 1:
  File "C:\Users\William\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 166:
    env.SConscript(item, exports="env")
  File "C:\Users\William\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\William\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\William\Documents\Marlin\Marlin\buildroot\share\PlatformIO\scripts\common-features-dependencies.py", line 135:
    install_features_dependencies()

thisiskeithb, 2020-07-20:
Manually delete the hidden `.pio` folder in addition to cleaning PIO (clicking the trash can icon on the bottom of VSCode).

I ran into the same issue on my builds.

rhapsodyv, 2020-07-20:
> 'CC' is not recognized as an internal or external command

We get the current compiler from the platformio env, but look like it don't have a valid one in windows, during the startup.

I will test it on windows.

teachingtechYT, 2020-07-20:
I have this error as well, VS code on Windows 10.
I have cleaned and deleted the .pio folder as Keith suggests.

thisiskeithb, 2020-07-20:
Seems this is a Windows-only issue since this was tested on Linux & macOS.

bill-orange, 2020-07-20:
@teachingtechYT Ditto.  Deleting .pio did not help.  I Should have mentioned in my original post that I am running Windows 10, as are you.

ellensp, 2020-07-20:
Confirmed broken on windows 10.

rhapsodyv, 2020-07-20:
Anyone can try this fix?

https://github.com/MarlinFirmware/Marlin/pull/18721

Azrael321, 2020-07-20:
> 
> 
> Anyone can try this fix?
> 
> #18721

Seems not to work

rhapsodyv, 2020-07-20:
> > Anyone can try this fix?
> > #18721
> 
> Seems not to work

Can you share me the output?

Azrael321, 2020-07-20:
> 
> 
> > > Anyone can try this fix?
> > > #18721
> > 
> > 
> > Seems not to work
> 
> Can you share the output?

`TypeError: sequence item 0: expected str instance, NoneType found:
  File "C:\users\doerr\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 166:
    env.SConscript(item, exports="env")
  File "C:\Users\doerr\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\doerr\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\Download\3D-Drucker\Firmware\Anet A6 MKS Gen L_BLTouch\Marlin-bugfix-2.0.x_20200720\buildroot\share\PlatformIO\scripts\common-features-dependencies.py", line 149:
    install_features_dependencies()
  File "D:\Download\3D-Drucker\Firmware\Anet A6 MKS Gen L_BLTouch\Marlin-bugfix-2.0.x_20200720\buildroot\share\PlatformIO\scripts\common-features-dependencies.py", line 42:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\doerr\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "D:\Download\3D-Drucker\Firmware\Anet A6 MKS Gen L_BLTouch\Marlin-bugfix-2.0.x_20200720\buildroot\share\PlatformIO\scripts\common-features-dependencies.py", line 142:
    load_marlin_features()
  File "D:\Download\3D-Drucker\Firmware\Anet A6 MKS Gen L_BLTouch\Marlin-bugfix-2.0.x_20200720\buildroot\share\PlatformIO\scripts\common-features-dependencies.py", line 131:
    cmd = ' '.join(cmd)`

ellensp, 2020-07-20:
I get the same error. 

I added a little debugging.
	cxx = search_compiler()
	print("cxx:")
	print(cxx)
and got 
cxx:
None

rhapsodyv, 2020-07-20:
Is there any output like this:
```
C:\Users\Victor\.platformio\packages\toolchain-gccarmnoneeabi\bin
arm-none-eabi-g++.exe -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-features-dependencies.h
```

I'm compiling with success in Windows 10 using the zip in this issue.

rhapsodyv, 2020-07-20:
Can you share the output of this:
```
print(env['ENV'])
```

You can put in anywhere in the `buildroot\share\PlatformIO\scripts\common-features-dependencies.py` file.

Azrael321, 2020-07-20:
> 
> 
> Can you share the output of this:
> 
> ```
> print(env['ENV'])
> ```
> 
> You can put in anywhere in the `buildroot\share\PlatformIO\scripts\common-features-dependencies.py` file.

`'Windows_NT', 'PATH': 'C:\\Users\\doerr\\.platformio\\packages\\tool-avrdude;C:\\Users\\doerr\\.platformio\\packages\\toolchain-atmelavr\\bin;C:\\Users\\doerr\\.platformio\\penv\\Scripts;C:\\Users\\doerr\\.platformio\\penv;C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\Program Files (x86)\\Intel\\Intel(R) Management Engine Components\\iCLS\\;C:\\Program Files\\Intel\\Intel(R) Management Engine Components\\iCLS\\;C:\\ProgramData\\Oracle\\Java\\javapath;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Program Files\\PuTTY\\;C:\\Program Files\\dotnet\\;C:\\Program Files (x86)\\Intel\\Intel(R) Management Engine Components\\DAL;C:\\Program Files\\Intel\\Intel(R) Management Engine Components\\DAL;C:\\Program Files (x86)\\Intel\\Intel(R) Management Engine Components\\IPT;C:\\Program Files\\Intel\\Intel(R) Management Engine Components\\IPT;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\LLVM\\bin;C:\\Program Files\\Intel\\WiFi\\bin\\;C:\\Program Files\\Common Files\\Intel\\WirelessCommon\\;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Users\\doerr\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\doerr\\AppData\\Local\\Programs\\Microsoft VS Code\\bin', 'PATHEXT': '.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC', 'PROCESSOR_ARCHITECTURE': 'AMD64', 'PROCESSOR_IDENTIFIER': 'Intel64 Family 6 Model 94 Stepping 3, GenuineIntel', 'PROCESSOR_LEVEL': '6', 'PROCESSOR_REVISION': '5e03', 'PROGRAMDATA': 'C:\\ProgramData', 'PROGRAMFILES': 'C:\\Program Files', 'PROGRAMFILES(X86)': 'C:\\Program Files (x86)', 'PROGRAMW6432': 'C:\\Program Files', 'PSMODULEPATH': 'C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules', 'PUBLIC': 'C:\\Users\\Public', 'PYTHONPATH': 'C:\\Users\\doerr\\.platformio\\penv\\Scripts\\platformio.exe;C:\\Users\\doerr\\.platformio\\python37\\python37.zip;C:\\Users\\doerr\\.platformio\\python37\\DLLs;C:\\Users\\doerr\\.platformio\\python37\\lib;C:\\Users\\doerr\\.platformio\\python37;c:\\users\\doerr\\.platformio\\penv;c:\\users\\doerr\\.platformio\\penv\\lib\\site-packages', 'SESSIONNAME': 'Console', 'SYSTEMDRIVE': 'C:', 'SYSTEMROOT': 'C:\\WINDOWS', 'TEMP': 'C:\\Temp', 'TMP': 'C:\\Temp', 'USERDNSDOMAIN': 'CSAE.LOCAL', 'USERDOMAIN': 'CSAE', 'USERDOMAIN_ROAMINGPROFILE': 'CSAE', 'USERNAME': 'Doerr', 'USERPROFILE': 'C:\\Users\\doerr', 'WINDIR': 'C:\\WINDOWS', 'AMD_ENTRYPOINT': 'vs/workbench/services/extensions/node/extensionHostProcess', 'APPLICATION_INSIGHTS_NO_DIAGNOSTIC_CHANNEL': 'true', 'ELECTRON_RUN_AS_NODE': '1', 'NO_PROXY': '127.0.0.1', 'PIPE_LOGGING': 'true', 'PLATFORMIO_CALLER': 'vscode', 'PLATFORMIO_IDE': '1.10.0', 'VERBOSE_LOGGING': 'true', 'VSCODE_CWD': 'C:\\Users\\doerr\\AppData\\Local\\Programs\\Microsoft VS Code', 'VSCODE_HANDLES_UNCAUGHT_ERRORS': 'true', 'VSCODE_IPC_HOOK': '\\\\.\\pipe\\e4e838c4148fa4cc504c4026214dd8c1-1.47.2-main-sock', 'VSCODE_IPC_HOOK_EXTHOST': '\\\\.\\pipe\\vscode-ipc-ee16cdfc-d04b-4cd7-aab7-0ca9423a5c0c-sock', 'VSCODE_LOGS': 'C:\\Users\\doerr\\AppData\\Roaming\\Code\\logs\\20200720T073831', 'VSCODE_LOG_STACK': 'false', 'VSCODE_NLS_CONFIG': '{"locale":"de","availableLanguages":{},"_languagePackSupport":true}', 'VSCODE_NODE_CACHED_DATA_DIR': 'C:\\Users\\doerr\\AppData\\Roaming\\Code\\CachedData\\17299e413d5590b14ab0340ea477cdd86ff13daf', 'VSCODE_PID': '1256', 'TERM_PROGRAM': 'vscode', 'TERM_PROGRAM_VERSION': '1.47.2', 'LANG': 'de_DE.UTF-8', 'COLORTERM': 'truecolor'})`

rhapsodyv, 2020-07-20:
Thanks! I could spot the difference.

Can you test the last push I did? Same branch.

Its a more robust search for the compiler path.

Azrael321, 2020-07-20:
> 
> 
> Thanks! I could spot the difference.
> 
> Can you test the last push I did? Same branch.
> 
> Its a more robust search for the compiler path.

Perfect, now everything is running fine!

ellensp, 2020-07-20:
Yes this looks to have fixed it.

rhapsodyv, 2020-07-20:
Great! Thanks for your help!

closed 2020-07-20 by thinkyhead