MarlinFirmware/Marlin #18722: [BUG] G33 P10 produces "Unknown command" on Delta [Marlin 2.0.5.4]

Ohmarinus, 2020-07-20:
### Bug Description

After upgrading to Marlin 2.0.5.4 I am no longer able to use G33 for Delta Auto Calibration. This worked on 2.0.5.3
Also, the EEPROM doesn't show it's values any longer when connecting to the machine with a host.

### My Configurations

[M2054_mar.zip](https://github.com/MarlinFirmware/Marlin/files/4947238/M2054_mar.zip)

### Steps to Reproduce

I have used the Delta example config that was downloaded from Github
1. Send: G33 P10

**Expected behavior:**

Delta Auto Calibration

**Actual behavior:** 

SENDING:G33 P10
echo:Unknown command: "G33 P10"

#### Additional Information

SKR 1.3
Piezo leveling (confirmed working)

GMagician, 2020-07-20:
g3 needs DELTA_AUTO_CALIBRATION

>   // uncomment to add G33 Delta Auto-Calibration (Enable EEPROM_SETTINGS to store results)
>  //#define DELTA_AUTO_CALIBRATION

ellensp, 2020-07-20:
The settings your referring to are only displayed on SERIAL_PORT not SERIAL_PORT_2 and you have SERIAL_PORT connected to a touch screen

closed 2020-07-20 by ellensp