MarlinFirmware/Marlin #18725: [BUG] G28 O does not function properly

CRCinAU, 2020-07-20:
### Bug Description
`G28 O` should home only when the printer is in a 'unknown' position - like after a motor has timed out, or the steppers shut down via `M84`.

`G28 O` should not perform a home operation when the printer is in a known state.

Currently with bugfix-2.0.x, `G28 O` does not perform a home under any circumstances.

### My Configurations
N/A - affects all boards.

### Steps to Reproduce
1. G28
2. M84
3. G28 O

thisiskeithb, 2020-07-20:
Per our discussion on Discord, I've confirmed the issue on one of my builds.

GMagician, 2020-07-20:
It could be a simple:
`if (!homing_needed() && parser.boolval('O')) {`
to
`if (!homing_needed() && parser.seen('O')) {`
in G28.cpp at line 221

CRCinAU, 2020-07-20:
Tested with:
```
  // Home (O)nly if position is unknown
  if (!homing_needed() && parser.seen('O')) {
    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("> homing not needed, skip");
    return;
  }
```

Test procedure: G28 / M84 / G28 O
Returns: ok, ok, ok - Single home is performed.

Test: G28 / G28 O
Returns: ok, ok - Single home is performed.

GMagician, 2020-07-20:
Ok, I had to abort my print....damn...I testd this code and it works as expected:
G28 home
G28 o doesn't home

M84 does nothing because I don't have HOME_AFTER_DEACTIVATE option enabled

CRCinAU, 2020-07-20:
I'm confused at the intent here.... M84 disables the steppers - which means that we are no longer 100% sure of where we are - as a shift might have happened, but we don't require homing to move again...

The only comments I can find for HOME_AFTER_DEACTIVATE are:
`//#define HOME_AFTER_DEACTIVATE  // Require rehoming after steppers are deactivated`

I interpret this as you can't move again until you home first... Which is not really a requirement - as I can use relative moves at my own risk, but I can't move to an absolute position... In the homing case, after an M84, we aren't able to guarantee any positional data - hence a `G28 o` should do a homing move.

Or, is the documentation (or my interpretation) a little out and `HOME_AFTER_DEACTIVATE` only mean that `homing_needed()` will return true on any stepper de-energise?

CRCinAU, 2020-07-20:
In fact, it looks like the expectation is exactly reversed here: https://github.com/MarlinFirmware/Marlin/issues/11271

GMagician, 2020-07-20:
HOME_AFTER_DEACTIVATE is used to "require" a home when stepper are de-energised

GMagician, 2020-07-20:
@CRCinAU #11271 has been fixed.

CRCinAU, 2020-07-20:
Correct - and the assumption is the exact opposite to the one I had...

Referring back to the Marlin documentation for G28, it states:

[O] | Optional. If the position is known then exit without homing. (1.1.9)

By definition, if a stepper(s) have been de-energised, then the position cannot be known. Therefore it seems that an M84 should still set the flag required to trigger a home via `G28 O`.

Otherwise, which bit of the documentation is incorrect? :|

GMagician, 2020-07-20:
G28 position is not known has to be considered as never homed or deenergized and difference is option above

thisiskeithb, 2020-07-20:
> Otherwise, which bit of the documentation is incorrect? :|

I left a comment in the now closed PR, but it’s not clear that `G28 O` will only work unless you have `HOME_AFTER_DEACTIVATE` enabled since there’s no mention of `HOME_AFTER_DEACTIVATE` anywhere on that page.

thinkyhead, 2020-07-20:
The `all_axes_homed` flags are set as soon as you do a complete `G28`, and theoretically they should never get cleared after that until you power off the machine. However, Marlin is an open system, and some function might mark an axis as "unhomed" for some as-yet-unspecified reason, even if `HOME_AFTER_DEACTIVATE` is not enabled. For this reason, `G28 O` exists.

So you can do this test:

With `HOME_AFTER_DEACTIVATE` disabled, boot up the machine and send `G28 O`. It will do a complete homing. Then send another `G28 O`. This time the machine is homed, so it does nothing.

closed 2020-07-20 by thinkyhead

thisiskeithb, 2020-07-20:
> With HOME_AFTER_DEACTIVATE disabled, boot up the machine and send G28 O. It will do a complete homing. Then send another G28 O. This time the machine is homed, so it does nothing.

Right, but if steppers timeout and you send another `G28 O`, it won’t home when it should since XYZ position is unknown.

Per the G28 page, the `G28 O` option says “ If the position is known then exit without homing. (1.1.9)”

CRCinAU, 2020-07-21:
> Per the G28 page, the `G28 O` option says “ If the position is known then exit without homing. (1.1.9)”, so with that logic, if the position is unknown, it shouldn’t exit and should home all axis.

This is my understanding also... by the very definition, when `M84` is issued, the position is then unknown. Therefore, a `G28 O` afterwards should trigger a home.

As a different test, the following `G28 O` does nothing:
G28
M84 S5
<wait 10 seconds>
G28 O

Without a timeout also causes no home:
G28
M84
G28 O

What does work is:
G28
G28 O

The last example only homes once.

The issue with that is if the steppers turn off due to either timeout, or specific instruction in end G-Code for instance, an ignored `G28 O` will mean that the printer can start moving with incorrect coordinates - which is in direct contradiction to what the documentation states for the O flag.

GMagician, 2020-07-21:
@CRCinAU it seems that 

> If the position is known

should be considered that
`#define HOME_AFTER_DEACTIVATE`: position is known after an home has been done and no M84 (or de-energization) occured
`//#define HOME_AFTER_DEACTIVATE`: position is known after an home has been done