MarlinFirmware/Marlin #18732: [BUG] Compilation error when trying to compile for Anycubic I3 Mega: AnycubicSerial' was not declared in this scope      AnycubicSerial.write(ch);

hub-jba, 2020-07-21:
### Bug Description

Compilation error when trying to compile for Anycubic I3 Mega: AnycubicSerial' was not declared in this scope      AnycubicSerial.write(ch);

I believe this has to do with the order of the FORCE_INLINE function in the anycubic_serial.h file. I think thinky rearranged the order to fit the conventions as part of my PR, and now it won't compile.

### My Configurations

Stock files out of the following folder in the Configurations repo default branch:
Configurations/config/examples/AnyCubic/i3 Mega/

### Steps to Reproduce

1. Checkout bugfix-2.0.x of Marlin
2. Checkout default branch of Configurations
3. overwrite configurations with config files from Configurations/config/examples/AnyCubic/i3 Mega
4. Try to build marlino with platform io
5. get this error:
In file included from Marlin/src/lcd/extui/lib/anycubic/anycubic_serial.cpp:38:0:
Marlin/src/lcd/extui/lib/anycubic/anycubic_serial.h: In function 'void AnycubicSerialprintPGM(const char*)':
Marlin/src/lcd/extui/lib/anycubic/anycubic_serial.h:136:5: error: 'AnycubicSerial' was not declared in this scope
     AnycubicSerial.write(ch);

**Expected behavior:** [What you expect to happen]
Successful build for AI3M.

**Actual behavior:** [What actually happens]
Compilation error:
In file included from Marlin/src/lcd/extui/lib/anycubic/anycubic_serial.cpp:38:0:
Marlin/src/lcd/extui/lib/anycubic/anycubic_serial.h: In function 'void AnycubicSerialprintPGM(const char*)':
Marlin/src/lcd/extui/lib/anycubic/anycubic_serial.h:136:5: error: 'AnycubicSerial' was not declared in this scope
     AnycubicSerial.write(ch);

hub-jba, 2020-07-21:
In file:
Marlin/src/lcd/extui/lib/anycubic/anycubic_serial.h
Need to move the following code block:

#ifdef UBRR3H
  extern AnycubicSerialClass AnycubicSerial;
#endif

to a part of the file before the following code block:

FORCE_INLINE void AnycubicSerialprintPGM(const char *str) {
  char ch=pgm_read_byte(str);
  while (ch) {
    AnycubicSerial.write(ch);
    ch=pgm_read_byte(++str);
  }
}

thinkyhead, 2020-07-21:
Patched in 2119e19bff

closed 2020-07-21 by thinkyhead