MarlinFirmware/Marlin #18733: [BUG] Enders 3 V2, Select Auto Bed Leveling on the controller will freeze the machine

fleek, 2020-07-21:
### Bug Description

Selecting the bed leveling option on the controller will freeze the machine if Autohome was not selected prior.

### My Configurations
[Marlin-bugfix-2.0.x-Ender3V2-BLTouchOn5Pin-18Jul2020.zip](https://github.com/MarlinFirmware/Marlin/files/4952997/Marlin-bugfix-2.0.x-Ender3V2-BLTouchOn5Pin-18Jul2020.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

#### Additional Information

G29 expects a prior G28.  However, if the menu option for G29 was not chosen in sequence after a G28 (Autohome), it will freeze the machine.

To check if Autohome was being done and if not issue G28 command prior to G29.

fleek, 2020-07-24:
I traced the function to  dwin.cpp

`/* Leveling */
void HMI_Leveling(void) {
  Popup_Window_Leveling();
  DWIN_UpdateLCD();
  queue.inject_P(PSTR("G29"));
}`

Currently it is only injecting G29, however, if there is no prior G28 being done, it will result in a hang of the machine.
Is there a way to check if G28 was performed before?  Otherwise could a forceful G28 be injected prior to G29?

Will there be a issue if the code is changed to

`/* Leveling */
void HMI_Leveling(void) {
  Popup_Window_Leveling();
  DWIN_UpdateLCD();
  queue.inject_P(PSTR("G28"));
  queue.inject_P(PSTR("G29"));
}`

Sorry, I am not quite familiar with the marlin code, so I need to ask before I do something silly.

CRCinAU, 2020-07-26:
@fleek - This is one of the patches I finalised last week but didn't get a chance to submit... 

I suggest: `G28 O\nG29\nM420 S1\n` - otherwise the machine doesn't have a way to enable to grid to be used.

fleek, 2020-07-26:
M420 S1 is to enable the use of the grid after a G28 because G28 disables the grid usage.  However, when a G29 is issued, the grid usage is automatically enabled.  G28 O\nG29\nM500 would be more appropriate as it will save the probed grid to EEPROM.

PR #18767 (2020-07-25): Ender 3 V2: Inject G28 before G29
### Requirements

When the bed leveling option is selected on the controller, if there is no prior Autohoming is done, the machine will hang.

### Description

Forcefully inject G28 prior to G29 to ensure that the bed leveling function is performed and controller returns to normal.

### Benefits

Fixes #18733
closed 2020-07-26 by fleek