MarlinFirmware/Marlin #18739: [BUG] Marlin Bugfix fails to compile after #18724 and #18730 updates

bill-orange, 2020-07-21:
Bug Description: After updates #18724 and #18730 Marlin fails to compile due to unsatisfied dependencies.  VSCODE was used in a Windows 10 environment.  Here is the error:

> Marlin\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp:62:10: fatal error: U8glib.h: No such file or directory
> 
> ****************************************************************
> * Looking for U8glib.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:U8glib.h"
> * Web  > https://platformio.org/lib/search?query=header:U8glib.h
> *
> ****************************************************************
> 
>    62 | #include <U8glib.h>
>       |          ^~~~~~~~~~
> compilation terminated.
> *** [.pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o] Error 1
> Marlin\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp:82:10: fatal error: U8glib.h: No such file or directory
> 
> ****************************************************************
> * Looking for U8glib.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:U8glib.h"
> * Web  > https://platformio.org/lib/search?query=header:U8glib.h
> *
> ****************************************************************
> 
>    82 | #include <U8glib.h>
>       |          ^~~~~~~~~~
> compilation terminated.
> *** [.pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o] Error 1

My Configurations: Attached  
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4954104/Marlin.zip)

Steps to Reproduce

1. Do a PlatformIO : Clean (just to be on the safe side)
2. Start Compilation and note immediate error

Expected behavior: Successful resolution of dependencies and compilation 

Actual behavior: Failure to resolve dependencies and failure to compile

Additional Information:  A brief look in Libraries shows that U8glib-HAL is present which appears to also contain U8glib-ARM.  These apparently do not resolve the dependency.

Lord-Quake, 2020-07-21:
I just downloaded https://github.com/MarlinFirmware/Marlin/commit/2119e19bff477be2bcefcf9b7f50ff60fdccee32 and was able to compile the code successfully.

bill-orange, 2020-07-21:
@Lord-Quake 2119e19 is the same version that would not compile for me.  It is likely that the dependency was already resolved in your environment.  I was not so lucky.  This problem has not cropped up in prior versions.

p3p, 2020-07-21:
Can you show the top of the compiler log where it builds the dependency graph.

rhapsodyv, 2020-07-21:
U8glib only will be installed and used if `HAS_GRAPHICAL_LCD` has been defined. Even if you already have U8glib downloaded in the lib_deps folder, it won't be found by platformio, because it will be explicit ignored.

So I think one of two thing might be happening:
 1. If you are using a U8glib "based" LCD, it may not be defining `HAS_GRAPHICAL_LCD`. It can be happens, as before those last changes, there aren't no explicit dependency management. What is you current selected LCD?

 2. If you not using a U8glib "based" LCD, some of your files may be referencing U8glib.h, but now using. Its another common problem will rise now, because before today, all the references were there... To fix, just include U8glib.h if `HAS_GRAPHICAL_LCD` is enabled:
```
#if ENABLED(HAS_GRAPHICAL_LCD)
  #include < U8glib.h>
#endif

Are you using the last bugfix? There're more than one PR about the new build that was just merged.

I will test your configs too.

Lord-Quake, 2020-07-21:
Using his configs:

```
> Executing task: C:\Users\Leo\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.3 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.4
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...     
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0    
|-- <Adafruit NeoPixel> 1.5.0
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\LCD_pin_routines.c.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Marlin\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp:62:10: fatal error: U8glib.h: No such file or directory

****************************************************************
* Looking for U8glib.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:U8glib.h"
* Web  > https://platformio.org/lib/search?query=header:U8glib.h
*
****************************************************************

   62 | #include <U8glib.h>
      |          ^~~~~~~~~~
compilation terminated.
Marlin\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp:82:10: fatal error: U8glib.h: No such file or directory

****************************************************************
* Looking for U8glib.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:U8glib.h"
* Web  > https://platformio.org/lib/search?query=header:U8glib.h
*
****************************************************************

   82 | #include <U8glib.h>
      |          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o] Error 1
Marlin\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp:62:10: fatal error: U8glib.h: No such file or directory

****************************************************************
* Looking for U8glib.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:U8glib.h"
* Web  > https://platformio.org/lib/search?query=header:U8glib.h
*
****************************************************************

   62 | #include <U8glib.h>
      |          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o] Error 1
====================================================================================================================== [FAILED] Took 4.23 seconds ====================================================================================================================== 

```

ellensp, 2020-07-21:
@bill-orange please remove the < and > characters from your last post as githug sees these as tags and hides version numbers in them

ellensp, 2020-07-21:
lpc1768/9 needs special p3p modified version of Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/1.5.0.zip

bill-orange, 2020-07-21:
@ellensp Why would it need it now and not before?  I am not doubting you but it just does not seem right.  

How would it be installed?

rhapsodyv, 2020-07-21:
Looks like REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER isn’t defining HAS_GRAPHICAL_LCD, but depends on U8glib.

ellensp, 2020-07-21:
Windows its not including TMCStepper or U8glib-HAL
On linux I get  
|-- Servo 1.0.0
|-- LiquidCrystal 1.0.0
|-- Adafruit NeoPixel 1.5.0
|-- TMCStepper 0.7.1
|-- U8glib-HAL 0.4.1

rhapsodyv, 2020-07-21:
@bill-orange 
try add it in the `[features]` section to see if it will work (I will test it soon):
```
[features]
U8GLIB_ST7920   = U8glib-HAL@0.4.1
                  src_filter=+<src/lcd/dogm>
```

p3p, 2020-07-21:
Sorry took me a second to get set up on windows to test this, your compile logs don't appear to be correct for the latest bugfix, when I build you see the feature detection script running as it should, I don't see it at all on yours.
```
> Executing task: C:\Users\P3P\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
arm-none-eabi-g++.exe -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-features-dependencies.h
Adding lib_deps for HAS_TRINAMIC_CONFIG... 
Adding lib_deps for HAS_GRAPHICAL_LCD... 
Adding src_filter for HAS_GRAPHICAL_LCD...
Adding lib_deps for HAS_LCD_MENU...
Adding src_filter for HAS_LCD_MENU...
Ignoring libs:  ['ESP32SSDP', 'Arduino-L6470', 'SailfishLCD', 'MKS-LittlevGL', 'TMC26XStepper', 'arduinoWebSockets', 'ESP Async WebServer', 'SlowSoftI2CMaster', 'AsyncTCP', 'Adafruit MAX31865 library', 'LiquidTWI2', 'ESP3DLib']
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.3 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.4
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <Adafruit NeoPixel> 1.5.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.1
Building in release mode
```

rhapsodyv, 2020-07-21:
> Sorry took me a second to get set up on windows to test this, your compile logs don't appear to be correct for the latest bugfix, when I build you see the feature detection script running as it [...]

Hum... Bingo!! I didn't see it!! But yes, you are correct.

This lines must appear with last commits:
```
arm-none-eabi-g++.exe -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-features-dependencies.h
```
and
```
Ignoring libs:  
```

p3p, 2020-07-21:
Ah, your `platofrmio.ini` file is not correct, many modifications from bugfix, so the scripts are not running correctly.

rhapsodyv, 2020-07-21:
> Ah, your `platofrmio.ini` file is not correct, many modifications from bugfix, so the scripts are not running correctly.

If those lines want showing, or isnt the lasts commits, or the user env was edited and isn't including the common.extra_scripts.

Lord-Quake, 2020-07-21:
Yep, I fell for it too. Should have checked that he is using the latest platformio.ini :-/

My test complies correctly now....

p3p, 2020-07-21:
@Lord-Quake The platformio.ini file was the first thing I checked, but I fell for the trap that the top lines looked correct having the new extra_script definitions, should have looked closer.

That looks fine?, Unless you have a drive with a `firmware.cur" file in the root it can't autodetect the upload directory.

Lord-Quake, 2020-07-21:
Must be something to with "default_envs = LPC1768". I've never compiled this before.

bill-orange, 2020-07-21:
@rhapsodyv  , @Lord-Quake   , @p3p  I am not sure how my PlatformIO could be outdated.  I thought Github Desktop Merge into Current Branch took care of those details.  That's neither here nor there, I suppose.  

In VSCODE I deleted the contents of the old PlatformIO content and pasted in the new content out of the Marlin Github site and changed my default_envs to LPC1768. Crude but effective.

Now I can successfully compile.

Do you suppose something needs to be added to Sanity Check for this?

p3p, 2020-07-21:
There isn't really a good way to detect an out if date or corrupted/modified platformio.ini, and if we did add a method for that it would probably use the scripts that were disabled by the merge corruption in your case so wouldn't have helped.

You have to be careful using git merges, I've never used github desktop so don't know what it does but you don't really want to merge the remote bugfix branch into your local copy to update, you should have a branch for your config and when you want to update you reset your local bugfix branch to origin then rebase your config branch on that.. but if your unfamiliar with git just downloading a new zip may be easier.

ellensp, 2020-07-21:
Resolved. Closing

closed 2020-07-21 by ellensp

thisiskeithb, 2020-07-21:
I'm not sure how easy rebasing in GitHub Desktop is, but it's literally drag and drop in GitKraken if you want to stick with a GUI client.

@bill-orange: You'll likely want to start with a fresh copy of `bugfix-2.0.x` to make sure there aren't other missing files/changes.