MarlinFirmware/Marlin #18740: [BUG] LCD FYSETC_MINI_12864 v1.2 only shows boot logo on SKR v1.4 Turbo

CaptainJackas, 2020-07-21:
### Bug Description

After switching my MKS Gen L for a BTT SKR v1.4 Turbo I can't get my LCD to work. It now shows the Marlin boot logo (which looks just fine), but then the menu pops up with almost no contrast at all which makes it basically invisible.

### My Configurations

[](https://we.tl/t-IsJdv4Gd3l)

### Steps to Reproduce

Nothing special, I just made the following adjustments to the config files (as found on the FYSETC wiki):

1. #define FYSETC_MINI_12864_1_2 // Configration.h
2. #define RGB_LED //Configration.h
3. #define LED_CONTROL_MENU //Configration_adv.h

**Expected behavior:** Show menu in same contrast as boot logo

**Actual behavior:** Menu almost invisible due to lack of contrast.

ellensp, 2020-07-21:
I believe there is a contrast menu on the lcd,  
or try M250

CaptainJackas, 2020-07-21:
As I couldn't see the LCD your first suggestion wasn't possible but M250 C255 fixed it! Thanks a lot!

closed 2020-07-21 by CaptainJackas