MarlinFirmware/Marlin #18742: PREHEAT_2_* seems broken on bugfix-2.0.x

shawe, 2020-07-21:
### Bug Description

On printer menu always appear text PREHEAT_1_LABEL also for PREHEAT_2_LABEL, this seems to be only reproducible on bugfix-2.0.x for at least a week. The temperatures are changing according to PREHEAT_1 or PREHEAT_2, but text messages aren't the correct.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

For me seems to not be related to a configuration problem.

[MKS Robin E3 BMG.zip](https://github.com/MarlinFirmware/Marlin/files/4956924/MKS.Robin.E3.BMG.zip)

### Steps to Reproduce

1. Compile
2. Update the firmware printer
3. Go to: Temperature an scroll down, Preheat PLA, scroll a little more and there are again a Preheat PLA
4. There is not PTG (or ABS), only PLA

**Expected behavior:** [What you expect to happen]

1. Compile
2. Update the firmware printer
3. Go to: Temperature an scroll down, Preheat PLA, scroll a little more and there are Preheat PTG (or ABS)

**Actual behavior:** [What actually happens]

1. Compile
2. Update the firmware printer
3. Go to: Temperature an scroll down, Preheat PLA, scroll a little more and there are again a Preheat PLA
4. There is not PTG (or ABS), only PLA

On stable version 2.0.x:

1. Compile
2. Update the firmware printer
3. Go to: Temperature an scroll down, Preheat PLA, scroll a little more and there are Preheat PTG (or ABS)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

[image]

thisiskeithb, 2020-07-22:
Replace `Marlin/src/lcd/language/language_es.h` with the `language_es.h` from the .zip: 
[language_es.h.zip](https://github.com/MarlinFirmware/Marlin/files/4957138/language_es.h.zip)

Or you can manually edit `language_es.h` & change the following:

https://github.com/MarlinFirmware/Marlin/blob/2af216cf828ee2276769b565896cc7a374e8210f/Marlin/src/lcd/language/language_es.h#L72-L88

to:

```cpp
#if PREHEAT_COUNT
    PROGMEM Language_Str MSG_PREHEAT_1                     = _UxGT("Precal. ") PREHEAT_1_LABEL;
    PROGMEM Language_Str MSG_PREHEAT_1_H                   = _UxGT("Precal. ") PREHEAT_1_LABEL " ~";
    PROGMEM Language_Str MSG_PREHEAT_1_END                 = _UxGT("Precal. ") PREHEAT_1_LABEL _UxGT(" Fusor");
    PROGMEM Language_Str MSG_PREHEAT_1_END_E               = _UxGT("Precal. ") PREHEAT_1_LABEL _UxGT(" Fusor ~");
    PROGMEM Language_Str MSG_PREHEAT_1_ALL                 = _UxGT("Precal. ") PREHEAT_1_LABEL _UxGT(" Todo");
    PROGMEM Language_Str MSG_PREHEAT_1_BEDONLY             = _UxGT("Precal. ") PREHEAT_1_LABEL _UxGT(" Cama");
    PROGMEM Language_Str MSG_PREHEAT_1_SETTINGS            = _UxGT("Precal. ") PREHEAT_1_LABEL _UxGT(" Ajuste");

    PROGMEM Language_Str MSG_PREHEAT_M                    = _UxGT("Precal. $");
    PROGMEM Language_Str MSG_PREHEAT_M_H                  = _UxGT("Precal. $ ~");
    PROGMEM Language_Str MSG_PREHEAT_M_END                = _UxGT("Precal. $ Fusor");
    PROGMEM Language_Str MSG_PREHEAT_M_END_E              = _UxGT("Precal. $ Fusor ~");
    PROGMEM Language_Str MSG_PREHEAT_M_ALL                = _UxGT("Precal. $ Todo");
    PROGMEM Language_Str MSG_PREHEAT_M_BEDONLY            = _UxGT("Precal. $ Cama");
    PROGMEM Language_Str MSG_PREHEAT_M_SETTINGS           = _UxGT("Precal. $ Ajuste");
 #endif
```

closed 2020-07-22 by thinkyhead

thisiskeithb, 2020-07-22:
The fix has now been merged into `bugfix-2.0.x`, so you don't have to edit files/replace text as long as you pull down a fresh copy.

shawe, 2020-07-22:
This was so fast ;)

I don't tested yet, but yes, seems to be this the problem because in other languages are the same.

Yesterday was too late for me when I see the problem, for try to look a little for the problem and suggest anything, and while I'm sleeping you found it and solve it.

PR #18743 (2020-07-22): Fix Spanish Preheat Options
### Requirements

`LCD_LANGUAGE es` (Spanish language)

### Description

Spanish preheat options were reversed, so everything past `PREHEAT_1` wouldn't display/work correctly.

### Benefits

Preheat options now display the correct filament name/type & preheat to the correct temps.

### Related Issues

Fixes #18742