MarlinFirmware/Marlin #18747: [FR] Purge after filament load

Thijsvanzwam, 2020-07-23:
Could it possible to have the nozzle do a purge of filament somewhere in a corner and then resume printing instead of doing it on the print it self? 
It now poops a blob of pla at the spot it resumes after a filament change which gives a bump in the print.
it would be nice to just do a small introline in the right corner (left is most used for introline at print start)and then resume the print.

Liamnerf, 2020-07-31:
this already exists. you need to have advanced pause and nozzle park enabled.

Thijsvanzwam, 2020-07-31:
I have both enabled. 
What is does is that it continues the print by moving the nozzle to the point it stopped and purges the filament there before continuing the print movements.
I have FILAMENT_CHANGE_FAST_LOAD_LENGTH and ADVANCED_PAUSE_PURGE_LENGTH   both set to 0

Liamnerf, 2020-08-01:
> I have both enabled.
> What is does is that it continues the print by moving the nozzle to the point it stopped and purges the filament there before continuing the print movements.
> I have FILAMENT_CHANGE_FAST_LOAD_LENGTH and ADVANCED_PAUSE_PURGE_LENGTH both set to 0

Those values are unrelated, you need to have resume prime (or prime on resume, whatever it’s called) set to 0.

LittleNyanCat, 2020-08-14:
This feature already exists, I use it. Search for #define NOZZLE_PARK_POINT in your Configuration.h. You can define the X & Y positions as well as how much to raise the Z axis.

qwewer0, 2021-01-01:
I think this can be closed, as we can purge on the parking spot.

slowbro, 2021-07-03:
Agreed, this is already a feature & can be closed.

thisiskeithb, 2021-07-03:
As noted above, this feature exists.

Simply enable `NOZZLE_PARK_FEATURE ` and define your preferred `NOZZLE_PARK_POINT` (off the build area for example) in order to avoid purging directly on your print.

closed 2021-07-03 by thisiskeithb