MarlinFirmware/Marlin #18755: [BUG] PID tuning works with external mosfet and re-assigned pins, but temperature thereafter does not rise

reefwarrior, 2020-07-23:
### Bug Description

### My Configurations

Included the .zip file. --- I have a MKS GEN 1.4 and have remapped the pins used to 'D7' as the original 'D8' mosfet has blown. I am using an external mosfet connected to D7.

### Steps to Reproduce

1. M303 E-1 C8 S60
2. When step 1 has finished, and all PID values have been saved/loaded again, then step 3.
3. M140 S60

**Expected behavior:** Heat bed is supposed to heat up.

**Actual behavior:** Heat bed does not heat up. In the M303, it heated up and was calibrated. After that it does not heat up.

#### Additional Information

The LED next to the mosfet used for D7, flashes faint and EXTREMELY fast and faint when doing the S140. For the M303 it flashes slower and brighter.
[configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/4966005/configuration_files.zip)

oldman4U, 2020-07-23:
Hi.

You are not the first to report issues with PID tuning of the bed and i would recommend not to use PID for the bed at all. It is not worth, because the bed is able to keep the temp. stable without it.

My 2 cents

oldman4U, 2020-07-23:
It likely does not heat, because the values from the PID tuning are bad. Turn PID for the bed off and try to heat it again.

reefwarrior, 2020-07-23:
Just tried, problem still persists.

oldman4U, 2020-07-23:
Ok. Sorry then i can not help you.

reefwarrior, 2020-07-23:
OK, problem has been resolved between V1.1.9 and the latest V2.0 - flashed, and it is running! :)

closed 2020-07-23 by reefwarrior