MarlinFirmware/Marlin #18756: [BUG] (E0 Fan not activating at 225 PWM)

dcwalmsley, 2020-07-23:
I have a custom built 3D Printer which incorporates 2x color mixing (2-in-1) assemblies fixed mounted to a single X-carrier.  The two units are BTT units and I have them both wired to a Azteeg X3 Pro board.  The concern I have with todays latest bug-fix-2.0 as well as all previous builds is a peculiar issue with Extruder 0.  

Both extruders are set to active at 75C with 225 PWM.  Extruder 1 works just fine and activates at roughly 75% fan operation but Extruder 0 will not spin.  If I change it in Configuration_h.adv line 463 to 255 PWM, both extruder fans will run at 100%.  I swapped out twice both fans and each are from the same manufacture batch and I get the exact same result.  I believe this is an issue or worse a bug.

I've uploaded the marlin folder for your review and ask for help determining if I screwed this up or is a an issue/bug.

Thanks for all you folks do.

V/R,
Doug W

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4968313/Marlin.zip)

Bob-the-Kuhn, 2020-07-23:
Is this correct?
```
  PWM setting for reliable startup

        EXTRUDER
  FAN |  0  |  1
  ----|-----|-------
   A  | 225 | 255 
   B  | 225 | 255
```

dcwalmsley, 2020-07-23:
Unclear on your question.  Where did you get these values?   I set up line 
423 on Configuration_adv.h is the only place I set at PWM 225.  Lines 423 & 
424 call out the fan pins.

I also feel that thete needs to be two PID values in Config.h vice a single 
one.  I know that I can use M303 for both extruders but I feel it also 
needs to reflect the PID value for each additional extruder.

~Doug

Bob-the-Kuhn, 2020-07-24:
Here's some code that adds a kick start time to the auto fans.

I have not been able to test it.  It does compile without error.

1. Add the following to **Configuration_adv.h**.  Adjust the time as needed
#define AUTO_FAN_KICKSTART_TIME 100  //run fan at full speed for xx ms to ensure reliable startup

2. In **temperature.cpp**,  replace the **checkExtruderAutoFans** section with the following:
[checkExtruderAutoFans.txt](https://github.com/MarlinFirmware/Marlin/files/4969708/checkExtruderAutoFans.txt)

---

I see that you are running an AZTEEG_X3_PRO and TMC2130s.  My first controller was an AZTEEG_X3_PRO and A4988s.  My current setup is a Big Tree Tech SKR_PRO_V1_1 and TMC5160s.  I you ever replace things then consider these items.

dcwalmsley, 2020-07-24:
I added "#define AUTO_FAN_KICKSTART_TIME 100 //run fan at full speed for xx
ms to ensure reliable startup" in line 466 just below "#define
CHAMBER_AUTO_FAN_SPEED 255".

Under temperature.cpp I modified lines 468, 670, and 1077 by adding
"checkExtruderAutoFans.txt" in place of *checkExtruderAutoFans.*

*This was the compiling error I received.*
*-----------------------------------------------------------------------------------------*

*Marlin\src\module\temperature.cpp: In static member function 'static void
Temperature::PID_autotune(const float&, heater_ind_t, int8_t, bool)':*
*Marlin\src\module\temperature.cpp:468:35: error: request for member 'txt'
in 'Temperature::checkExtruderAutoFans', which is of non-class type
'void()'*
*             checkExtruderAutoFans.txt();*
*                                   ^*
*Marlin\src\module\temperature.cpp: At global scope:*
*Marlin\src\module\temperature.cpp:670:42: error: expected initializer
before '.' token*
*   void Temperature::checkExtruderAutoFans.txt() {*
*                                          ^*
*Marlin\src\module\temperature.cpp:121:32: warning: 'heater_ttbl_map'
defined but not used [-Wunused-variable]*
*     static const temp_entry_t* heater_ttbl_map[HOTENDS] =
ARRAY_BY_HOTENDS(HEATER_0_TEMPTABLE REPEAT_S(1, HOTENDS, NEXT_TEMPTABLE));*
*                                ^*
**** [.pio\build\mega2560\src\src\module\temperature.cpp.o] Error 1*

dcwalmsley, 2020-07-24:
Deleted previous post.

I edited the two files, "Configuration_adv.h" and "temperature.cpp" but receiving the following compiling error.

Marlin\src\module\temperature.cpp: In static member function 'static void Temperature::checkExtruderAutoFans()':
Marlin\src\module\temperature.cpp:707:39: error: 'auto_FAN_KICKSTART_TIME' was not declared in this scope
           auto_fan_kick_end[P] = ms + auto_FAN_KICKSTART_TIME; \
                                       ^
Marlin\src\module\temperature.cpp:738:19: note: in expansion of macro '_UPDATE_AUTO_FAN'
           case 0: _UPDATE_AUTO_FAN(0, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
                   ^
Marlin\src\module\temperature.cpp:707:39: error: 'auto_FAN_KICKSTART_TIME' was not declared in this scope
           auto_fan_kick_end[P] = ms + auto_FAN_KICKSTART_TIME; \
                                       ^
Marlin\src\module\temperature.cpp:741:19: note: in expansion of macro '_UPDATE_AUTO_FAN'
           case 1: _UPDATE_AUTO_FAN(1, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
                   ^
*** [.pio\build\mega2560\src\src\module\temperature.cpp.o] Error 1**
**

Attached are the two files. 
[two files of interest.zip](https://github.com/MarlinFirmware/Marlin/files/4973335/two.files.of.interest.zip)

dcwalmsley, 2020-07-27:
Bob,

Where does "#define AUTO_FAN_KICKSTART_TIME 100" go to?   There is no reference to this in temperature.cpp.

Bob-the-Kuhn, 2020-07-27:
Maybe the 100ms kick start time is too short.  Try changing the time from 100 to 2000.

Maybe my code is foobar.  Try setting the time to something really long (5000).  Does the E0 fan spin up and the die after 5 seconds? 

If you swap the E0 and E1 fans, does the behavior follow the fans?

---

Why is the 225 important?

This sounds like a solution to a problem.  Maybe there is another solution to the problem.

I'm just going to throw out some changes I've made to solve prioblems with keeping the extruder at the desired temperature.  
- Lower volume fan.
- 40W heater instead of the usual 30W.
- Reduced the speed on the part cooling fan.
- Went with a 4 pin PWM fan (two power, PWM logic level control input, RPM output)  [example PWM fan](https://www.amazon.com/Noctua-Cooling-Blades-Bearing-NF-A4x10/dp/B009NQLT0M/ref=sr_1_1_sspa?dchild=1&keywords=pwm+12V+fan+40mm&qid=1595877535&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFSSllFUDBVVllPQkcmZW5jcnlwdGVkSWQ9QTA3OTY0NzkzUk1IUzFXRlo0SUs0JmVuY3J5cHRlZEFkSWQ9QTAwMjE2NDkxQVlYNkpSRTUzUjdUJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ== )  [PWM interface](https://en.wikipedia.org/wiki/Computer_fan_control)
- Partially blocked the fan openning.

Bob-the-Kuhn, 2020-07-27:
My code is foobar.   Everything should be **AUTO_FAN_KICKSTART_TIME** but there is one place that says **auto_FAN_KICKSTART_TIME.**

Try this:
[checkExtruderAutoFans (1).txt](https://github.com/MarlinFirmware/Marlin/files/4984090/checkExtruderAutoFans.1.txt)

dcwalmsley, 2020-07-27:
Bob,

You are right with 225 being a solution to a problem.  With two 40 
wattextruders sitting side by side opposed, I noticed the fans at full 
speed was dumping to much air onto the heater block which I believe is 
having a hard time with PID tuning and E0 failing when I try to get temp 
above 225C.

My theory is that that E1 is working fine as that fan is rotating and 
pushing air upward and not down onto heater block the way E0 is performing. 
 Therefore, E1 PID tunes and heats up just fine.

So my next process (not optimal) is to place a piece of thermal material 
between heater blocks and heatsink to see if this is the issue with my design.

If this works, then I know I need to be able to set different fan speeds 
for each extruder to be able to fine tune the cooling process.  The thermal 
seperator was never considered as a perminent solution for overly efficient 
fans.

On a side note, I was able to get your code working although both fans 
slowly come up to 255 PWM regardless what I set for speed in 
Configuration_adv.h.

I will upload the temperature.cpp file this evening for your review.

I did download 2.0.6 but when I implemented your code two thing didn't 
compile until I corrected them.

1. There are 8 cases which list E0 - E7.  When compiling the "E" in front 
of the numeral caused it to fail.  EE0 or something like that.

2. Your use of lower case "auto_FAN_KICKSTART_TIME" kicked a compile error. 
 I changed it to"AUTO_FAN_KICKSTART_TIME".

After there two areas were adjusted, everything compiled.

R, Doug Walmsley

dcwalmsley, 2020-07-27:
This is the code from my temperature.cpp which appears the same as your "checkExtruderAutoFans (1).txt" file
-------------------------------------------------------------------------------------
#if HAS_AUTO_FAN

  #define CHAMBER_FAN_INDEX HOTENDS

  void Temperature::checkExtruderAutoFans() {
    #define _EFAN(B,A) _EFANOVERLAP(A,B) ? B :
    static const uint8_t fanBit[] PROGMEM = {
      0
      #if HAS_MULTI_HOTEND
        #define _NEXT_FAN(N) , REPEAT2(N,_EFAN,N) N
        RREPEAT_S(1, HOTENDS, _NEXT_FAN)
      #endif
      #if HAS_AUTO_CHAMBER_FAN
        #define _CFAN(B) _FANOVERLAP(CHAMBER,B) ? B :
        , REPEAT(HOTENDS,_CFAN) (HOTENDS)
      #endif
    };

    uint8_t fanState = 0;
    HOTEND_LOOP()
      if (temp_hotend[e].celsius >= EXTRUDER_AUTO_FAN_TEMPERATURE)
        SBI(fanState, pgm_read_byte(&fanBit[e]));

    #if HAS_AUTO_CHAMBER_FAN
      if (temp_chamber.celsius >= CHAMBER_AUTO_FAN_TEMPERATURE)
        SBI(fanState, pgm_read_byte(&fanBit[CHAMBER_FAN_INDEX]));
    #endif

    #define __UPDATE_AUTO_FAN(P,D,A) do{                  \
      if (PWM_PIN(P##_AUTO_FAN_PIN) && A < 255)          \
        analogWrite(pin_t(P##_AUTO_FAN_PIN), D ? A : 0); \
      else                                               \
        WRITE(P##_AUTO_FAN_PIN, D);                      \
    }while(0)

    #if AUTO_FAN_KICKSTART_TIME > 0
      static millis_t auto_fan_kick_end[9] = { 0 };  // 8 extruders + chamber fan
      #define _UPDATE_AUTO_FAN(P,D,A)                         \
        if (D) {                       \
          millis_t ms = millis();                      \
          if (auto_fan_kick_end[P] == 0) {                  \
          auto_fan_kick_end[P] = ms + AUTO_FAN_KICKSTART_TIME; \
            __UPDATE_AUTO_FAN(E##P,D,255);                   \
          } else if (PENDING(ms, auto_fan_kick_end[f]))     \
            __UPDATE_AUTO_FAN(E##P,D,255);                   \
        } else {                   \
           auto_fan_kick_end[P] = 0;                   \
           __UPDATE_AUTO_FAN(E##P,D,A);}
    #else
      #define _UPDATE_AUTO_FAN(P,D,A) __UPDATE_AUTO_FAN(E##P,D,A)
    #endif

    uint8_t fanDone = 0;
    LOOP_L_N(f, COUNT(fanBit)) {
      const uint8_t realFan = pgm_read_byte(&fanBit[f]);
      if (TEST(fanDone, realFan)) continue;
      const bool fan_on = TEST(fanState, realFan);
      switch (f) {
        #if ENABLED(AUTO_POWER_CHAMBER_FAN)
          case CHAMBER_FAN_INDEX:
            chamberfan_speed = fan_on ? CHAMBER_AUTO_FAN_SPEED : 0;
            break;
        #endif
        default:
          #if ENABLED(AUTO_POWER_E_FANS)
            autofan_speed[realFan] = fan_on ? EXTRUDER_AUTO_FAN_SPEED : 0;
          #endif
          break;
      }

      switch (f) {
        #if HAS_AUTO_FAN_0
          case 0: _UPDATE_AUTO_FAN(0, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
        #endif
        #if HAS_AUTO_FAN_1
          case 1: _UPDATE_AUTO_FAN(1, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
        #endif
        #if HAS_AUTO_FAN_2
          case 2: _UPDATE_AUTO_FAN(2, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
        #endif
        #if HAS_AUTO_FAN_3
          case 3: _UPDATE_AUTO_FAN(3, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
        #endif
        #if HAS_AUTO_FAN_4
          case 4: _UPDATE_AUTO_FAN(4, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
        #endif
        #if HAS_AUTO_FAN_5
          case 5: _UPDATE_AUTO_FAN(5, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
        #endif
        #if HAS_AUTO_FAN_6
          case 6: _UPDATE_AUTO_FAN(6, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
        #endif
        #if HAS_AUTO_FAN_7
          case 7: _UPDATE_AUTO_FAN(7, fan_on, EXTRUDER_AUTO_FAN_SPEED); break;
#endif
        #if HAS_AUTO_CHAMBER_FAN && !AUTO_CHAMBER_IS_E
          case CHAMBER_FAN_INDEX: _UPDATE_AUTO_FAN(CHAMBER, fan_on, CHAMBER_AUTO_FAN_SPEED); break;
        #endif
      }
      SBI(fanDone, realFan);
    }
  }

#endif // HAS_AUTO_FAN

closed 2020-09-01 by github-actions[bot]

dcwalmsley, 2020-09-19:
Latest 2.0.6.1 and some tweaking, I have both extruders heating to desired settings.  

Closing this ticket.