MarlinFirmware/Marlin #18757: [BUG] Bed temp runaway when SD card is removed mid-print

Liamnerf, 2020-07-23:
### Bug Description

Just now I removed the SD card from my printer. the print halted in place, with the nozzle maintaining the set temperature. the Bed temperature, on the other hand, began to climb without my knowledge. When I came back to the printer, the bed temp was at 87C, despite it being set to 65C in the print file. Thermal runaway did not trigger. Using an SKR Mini E3 V1.2. 

### My Configurations

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/4968693/Archive.zip)

### Steps to Reproduce

1. Start print
2. wait till print is well underway
3. Remove SD card

Expected behaviour: For the print to stop, and for the nozzle and bed to maintain the set temperature.

Actual Behaviour: Bed temperature begins to rise without anything stopping it.

sjasonsmith, 2020-07-28:
Is the display still responsive when this happens, or is it all locked up?

Liamnerf, 2020-08-01:
> Is the display still responsive when this happens, or is it all locked up?

All locked up.

closed 2020-10-09 by github-actions[bot]