MarlinFirmware/Marlin #18760: [BUG] Cannot compile Marlin 2.0.5 on BTT SKR Mini E3 Dip

KyleLampkin, 2020-07-23:
I have a BTT SKR Mini E3 Dip and a BTT TFT 35. If I use the original firmare-tmc2130.bin file everything works as expected without some features I want to use. So I try to compile a new firmware with the SKR Mini E3 Dip master firmware but it will not compile without errors. So I try to use the latest marlin firmware or marlin bugfix and it compiles fine but then the touch screen will only work
in simulator mode.

boelle, 2020-07-23:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

KyleLampkin, 2020-07-24:
Sorry I didn't include those earlier, my bad. Here are the files you requested
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4969644/Configuration_adv.zip)

ellensp, 2020-07-24:
How about attaching some error logs?

ellensp, 2020-07-24:
Your config implies your using 2.0.5.4 if you haven't downloaded this very recently platformio grabs the latest version of libraries which no longer works with marlin. use  this platformio.ini https://raw.githubusercontent.com/MarlinFirmware/Marlin/2.0.x/platformio.ini and adjust your default_envs as needed.

Cant assist with bugfix issues as I dont have the hardware.

KyleLampkin, 2020-07-24:
Awesome  worked like a charm with that ini file. But I have to say just updating software shouldn't cause this problem, should be backward compatible I would think.

Thank you  ellensp for all the help, I would never have figured that out.

closed 2020-07-24 by KyleLampkin