MarlinFirmware/Marlin #18761: Undefined reference to BL24CXX when using CR10_STOCKDISPLAY

mindspawn, 2020-07-23:
### Bug Description

Linker Warning: Undefined reference to BL24CXX when using CR10_STOCKDISPLAY

Starting from commit f34024af3457d1d5e2f6b4d7fe83f5b00f8fb597 and beyond this linker error shows up and causes the Ender 3 Pro v1.5 config to hang during power on at the Marlin splash screen (prior commits work fine).

### My Configurations

Configurations\config\examples\Creality\Ender-3 Pro V1.5 (import-2.0.x branch commit 79343f5c22600834cd74124cb1fc68609e659a38 )

Compiled with: platformio run -e STM32F103RET6_creality

### Steps to Reproduce

Build the above configuration.

**Expected behavior:** No linker error. Firmware does not hang at power on.

**Actual behavior:** Linker warning and Firmware hangs display the splash screen at power on.

#### Additional Information

rhapsodyv, 2020-07-23:
Might be solved by: #18758 

Can you test it?

mindspawn, 2020-07-23:
> Might be solved by: #18758
> 
> Can you test it?

Glad to test it. Do you know when the fix will show up on the bugfix-2.0.x branch?

rhapsodyv, 2020-07-24:
You can test it downloading the thinkhead branch: thinkyhead:bf2_ender3_v2_eeprom_PR

mindspawn, 2020-07-24:
> You can test it downloading the thinkhead branch: thinkyhead:bf2_ender3_v2_eeprom_PR

I just built and tested from this branch and it works again.. wooohooo :) thanks a million guys :)

closed 2020-07-24 by mindspawn