MarlinFirmware/Marlin #18765: [BUG] SKR mini V2.0 error: eeprom was not declared in this scope

mrseeker, 2020-07-24:
### Bug Description

When compiling 2.0.5-bugfix - 8b637e4 for the board BTT_SKR_MINI_E3_V2_0
I get the following error:

```
Marlin\src\HAL\STM32F1\eeprom.cpp: In static member function 'static bool PersistentStore::write_data(int&, const uint8_t*, size_t, uint16_t*)':
Marlin\src\HAL\STM32F1\eeprom.cpp:49:14: error: 'eeprom_read_byte' was not declared in this scope
     if (v != eeprom_read_byte(p)) {
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_flash.cpp.o
              ^~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\eeprom.cpp:49:14: note: suggested alternative: 'pgm_read_byte'
     if (v != eeprom_read_byte(p)) {
              ^~~~~~~~~~~~~~~~
              pgm_read_byte
Marlin\src\HAL\STM32F1\eeprom.cpp:50:7: error: 'eeprom_write_byte' was not declared in this scope
       eeprom_write_byte(p, v);
       ^~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\eeprom.cpp:50:7: note: suggested alternative: 'gpio_write_bit'
       eeprom_write_byte(p, v);
       ^~~~~~~~~~~~~~~~~
       gpio_write_bit
Marlin\src\HAL\STM32F1\eeprom.cpp: In static member function 'static bool PersistentStore::read_data(int&, uint8_t*, size_t, uint16_t*, bool)':
Marlin\src\HAL\STM32F1\eeprom.cpp:65:17: error: 'eeprom_read_byte' was not declared in this scope
     uint8_t c = eeprom_read_byte((uint8_t*)pos);
                 ^~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\eeprom.cpp:65:17: note: suggested alternative: 'pgm_read_byte'
     uint8_t c = eeprom_read_byte((uint8_t*)pos);
                 ^~~~~~~~~~~~~~~~
                 pgm_read_byte
Marlin\src\HAL\STM32F1\eeprom.cpp: In static member function 'static size_t PersistentStore::capacity()':
Marlin\src\HAL\STM32F1\eeprom.cpp:74:45: error: 'E2END' was not declared in this scope
 size_t PersistentStore::capacity() { return E2END + 1; }
                                             ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom.cpp.o] Error 1
```

This file is not found in the official release of BTT, so I suspect that this file causes some issues with my configuration.

### My Configurations
https://www.dropbox.com/s/fn3nsenb4cl8de3/Marlin.zip?dl=0

### Steps to Reproduce
Compile in PlatformIO

**Expected behavior:** [What you expect to happen]
A successful compilation

**Actual behavior:** [What actually happens]
Error during compilation

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mrseeker, 2020-07-24:
I found out that #18088 does work for me with a small change in the configuration.

thisiskeithb, 2020-07-25:
I just tried building your configs linked above & the [SKR Mini E3 V2 configs](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0) from the Marlin Configurations repo using the latest `bugfix-2.0.x` and they both build without errors.

mrseeker, 2020-07-25:
I checked, built seems to be fixed. Closing.

closed 2020-07-25 by mrseeker