MarlinFirmware/Marlin #18766: [BUG] Configuration: LA vs. JD vs. S-Curve

yoflo, 2020-07-24:
Hello everyone,
First of all: Sorry, this is not a special bug. I just keep reading issues with problems hinting at combinations of LA, JD and s-curve accelerations. I hope this is the right place to ask:
What is the current situation with these features? Should they be combined or what combinations are working fine?

Thanks in advance.

ellensp, 2020-07-24:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-24 by ellensp

yoflo, 2020-07-24:
Hello,

Thanks for your reply, however the Marlin documentation does not adress these issues and the user-based platforms give mixed replies. 
In my opinion only the developers can give a validate statement to these questions. Is there a way to get a statement from the development team?