MarlinFirmware/Marlin #18768: [BUG] Motor moves discontinuously

Jianxuncui, 2020-07-24:
### Bug Description

I used marlin to control the z motor, at first few times, the motor moves smoothly, but after some movements, the movement becomes discontinuous.
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4973457/Configuration_adv.zip)

### My Configurations
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4973462/Configuration_adv.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-07-28:
Please update the remaining sections in the template. Exact steps to reproduce is important, even if it seems obvious to you.

This is a good example where a video may be helpful to demonstrate the problem. It isn't obvious to me what exactly "the movement becomes discontinuous" means.

Jianxuncui, 2020-07-28:
We replaced the electronic board and fix the issue.

closed 2020-08-17 by sjasonsmith