MarlinFirmware/Marlin #18776: [BUG] Sensorless homing on delta printer not simultaneously

CaptainJackas, 2020-07-25:
### Bug Description

When turning on sensorless homing the carriages of my delta do not home simultaneously anymore which causes them to crash when homing from low Z values. I've seen this question asked quite a few times for the past two years but from my understanding this is still not fixed.

### My Configurations

SKR v1.4 + TMC2209 v1.2 on a delta printer

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4975760/Marlin.zip)

### Steps to Reproduce

1. Turn on Sensorless homing on a delta printer
 
**Expected behavior:** Homing all three axis at the same time (like without sensorless homing)

**Actual behavior:** Each axis homes individually which, when coming from low Z values, results in crashing of the effector into the tower

#### Additional Information

There has been some suggestions in issue #10141 which indeed results in simultaneous homing but this is very unreliable as sometimes it works, sometimes it doesnt, and if it works it only correctly homes the Z axis.

sjasonsmith, 2020-07-28:
Can anybody else using sensorless on a delta confirm this behavior?

It's not a very common combination, so there are likely to be issues.

Kadah, 2020-08-11:
I ran in to this exact issue as well when I initially setup my current printer build on a release version of 2.0 (I don't remember which exactly). I switched to to the bugfix-2.0.x and that resolved it, so this bug might already be fixed.

This problem seemed really familiar when I ran in to it a few weeks ago. I think I this exact same issue years ago when I rebuilt an earlier version of this printer on 1.x, and then too switching to to the bugfix branch resolved it. That's purely an amusing coincidence.

boelle, 2020-08-11:
@CaptainJackas can you confirm what @Kadah just mentioned? ie try 
 [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) and see if it works

Kadah, 2020-08-24:
Suddenly mine stopped working tonight. It was still working earlier today, did some config changes, to enable eeprom and account for changed height due to new bed, and then it was homing each individually.

Did a quick recompile with only #define EEPROM_SETTINGS commented out and homing was working again. Something about eeprom breaks delta homing, even with defaults loaded and saved.

@CaptainJackas Try without EEPROM_SETTINGS enabled and see homes correctly.

closed 2020-09-23 by github-actions[bot]

sjasonsmith, 2020-09-23:
I'm not sure why this was marked stale and closed. If anybody is still interested in this just reply and I'll re-open it for you.