MarlinFirmware/Marlin #18777: [BUG] Compile error when HAS_LCD_MENU is false in MarlinUI::external_encoder

ellensp, 2020-07-25:
### Bug Description

In ultralcd.h external_encoder() is only defined if HAS_LCD_MENU is true.
But in ultralcd.cpp the function external_encoder() is created when HAS_LCD_MENU is true or false.

With the right Configuration file, this can trigger the following compile error.
"Marlin/src/lcd/ultralcd.cpp:760:35: error: no 'void MarlinUI::external_encoder()' member function declared in class 'MarlinUI'"

### My Configurations

[Break_marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/4975833/Break_marlin_config.zip)

### Steps to Reproduce

1. Obtain latest marlin bugfix 2.0.x
2. Use provided Configuration.h 
3. Watch it fail to compile.

**Expected behavior:**

It should compile

**Actual behavior:**

It errors with 
Marlin/src/lcd/ultralcd.cpp:760:35: error: no 'void MarlinUI::external_encoder()' member function declared in class 'MarlinUI'

rhapsodyv, 2020-07-25:
The cause:

The method is compiled when `AUTO_BED_LEVELING_UBL` is enabled. But is defined only when 
`  #if HAS_LCD_MENU && EITHER(AUTO_BED_LEVELING_UBL, G26_MESH_VALIDATION)` and `#if ENABLED(AUTO_BED_LEVELING_UBL)`

```
  #if HAS_LCD_MENU && EITHER(AUTO_BED_LEVELING_UBL, G26_MESH_VALIDATION)
    static bool external_control;
    FORCE_INLINE static void capture() { external_control = true; }
    FORCE_INLINE static void release() { external_control = false; }
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      static void external_encoder();
    #endif
```

We must match the same conditions in cpp and the header files.

rhapsodyv, 2020-07-25:
I did the cpp match the header in #18778

closed 2020-07-25 by thinkyhead